cesanta/mongoose #1291: Feature request: Allow functions to be declared `static`

kristjanvalur, 2021-05-26:
- My goal is:  To be able to include mongoose in a large project as a module, where another module has possibly included mongoose.

- Specific:  A complex project using external plugins, where both plugins use mongoose, get link errors.  This is a game using **UnrealEngine**, where marketplace plugins use mongoose. A complete game will link all plugin modules into a single executable and the existence of both mongoose implementations in source `.c` files causes problems.
- Suggestion:  Add a macro, e.g.  `MG_STATIC`, by default defined as empty, to the external function decarations.  Libraries/modules can elect to `#include "mongoose.c"` with this define, thus hiding the mongoose library from the linker
- Alternative:  Add a macro, e.g. `MG_PREFIX`, by default empty, which is token-prefixed to external functions, thus in effect allowing the user to define a namespace for mongoose.

I have control over one of the plugins and can modify the way that it uses Mongoose.  But for another, commercial plugin, the library is embedded in a `.lib` file that I have no control over.  I will manually fix the plugin which I have access to (with the `static` solution mentioned above`, but it would be useful for others to have a way of isolating a library, which is designed to be embeddable, from other modules.

cpq, 2021-05-27:
Your plugin system is broken - if it works the way you describe, then the plugin mechanism should be fixed, not mongoose adopt kludges.

That's a typical "dependency" thing, and if several plugins could use mongoose, make mongoose an independent plugin/library and let other depend on it.

"Namespacing" mongoose to fight with linking conflict is just plain wrong, and there is no way mongoose is going to help with it.

closed 2021-05-27 by cpq

kristjanvalur, 2021-05-28:
Thank you for your kind reply.
This is not _my_ plugin system, but one in a large and successful
commercial product.  It is also not something that is unique to the product
but a problem with C itself.
Let me explain:
1) Normally, plugins are compiled into their own DLLs, and they can operate
stand-alone.  Multiple such plugins can include Mongoose internally.
2) Uniquely, and because this is a game engine, for release, plugins and
engine are compiled into a monolithic executable.  In those cases, third
party libraries that also include mongoose (or any other fourth party
libraries) can run into linking conflicts.

This is just how C and linking works.

Now, I realized that for this particular use case, I can simply use C++
namespaces to fix it, and include both header and implementation in such a
namespace.  Other environments might not be so lucky.

My suggestion was intended to make your library more friendly to atypical
use-cases.  Adding an API macro like this would also allow adding other
linking directives such as DLL exporting and other such things, that
otherwise have to be manually added.  Maintaining some OS projects myself,
I usually respond positively to such suggestions.

Cheers, and happy coding.