cesanta/mongoose #1293: Websocket connection is not closed upon reception of reserved opcode

GlebPlekhotko, 2021-05-29:
- My goal is: Close connection if reserved opcode is received.
- My actions were: Receive 3 bytes long websocket frame [0x8B, 0x01, 0x55]
- My expectation was: MG_EV_CLOSE event in respective handler function;
- The result I saw: MG_EV_WS_MSG in respective handler function;
- My question is: According to the "RFC6455 — The WebSocket Protocol" in case of unknown opcode, connection must be closed (clause "5.2 Base Framing Protocol", page 29). Version I'm using (7.0) as well as the latest one (7.3) does not do that automatically. Though user may implement this feature "by hand" by analyzing "flags" field of the incoming frame and setting of "is_closing" flag in a particular connection, that doesn't seem reliable. How do you think, is it reasonable to embed this feature into the library itself?

# Environment

- mongoose version: 7.0
- custom changes?: No
- build environment:
  - OS: Windows 10;
  - Compiler/IDE: ARM GCC 9.2.1;
- relevant example (see https://github.com/cesanta/mongoose/tree/master/examples) is: —
- relevant documentation topic (see https://cesanta.com/docs/) is: —

cpq, 2021-05-29:
Please apply the following change and see if that fixes your issue:

```diff
diff --git a/mongoose.c b/mongoose.c
index 0bd177f..78cccfb 100644
--- a/mongoose.c
+++ b/mongoose.c
@@ -4364,6 +4364,8 @@ static void mg_ws_cb(struct mg_connection *c, int ev, void *ev_data,
       char *s = (char *) c->recv.buf + msg.header_len;
       struct mg_ws_message m = {{s, msg.data_len}, msg.flags};
       switch (msg.flags & WEBSOCKET_FLAGS_MASK_OP) {
+        case WEBSOCKET_OP_CONTINUE:
+          break;
         case WEBSOCKET_OP_PING:
           LOG(LL_DEBUG, ("%s", "WS PONG"));
           mg_ws_send(c, s, msg.data_len, WEBSOCKET_OP_PONG);
@@ -4372,15 +4374,17 @@ static void mg_ws_cb(struct mg_connection *c, int ev, void *ev_data,
         case WEBSOCKET_OP_PONG:
           mg_call(c, MG_EV_WS_CTL, &m);
           break;
+        case WEBSOCKET_OP_TEXT:
+        case WEBSOCKET_OP_BINARY:
+          mg_call(c, MG_EV_WS_MSG, &m);
+          break;
+        default:
+          // FALLTHROUGH. Per RFC6455, close conn when an unknown op is recvd
         case WEBSOCKET_OP_CLOSE:
           LOG(LL_ERROR, ("%lu Got WS CLOSE", c->id));
           mg_call(c, MG_EV_WS_CTL, &m);
           c->is_closing = 1;
-          return;
-        default: {
-          mg_call(c, MG_EV_WS_MSG, &m);
           break;
-        }
       }
       mg_iobuf_delete(&c->recv, msg.header_len + msg.data_len);
     }
```

cpq, 2021-07-14:
The above patch is integrated and should solve an issue.
Closing this - feel free to reopen if the issue persists.

closed 2021-07-14 by cpq