cesanta/mongoose #1298: Issues when compiling with C++ code (GCC11)

Harvie, 2021-06-14:
I've tried to embed mongoose to C++ project.
While C code is usualy considered to build with C++ compiler without any modifications, it turns out there are some exceptions to this rule.

**I beleive it would make sense to run all the tests both with C and C++ compilers to prevent such problems.**

In my case it happens that C++ is not happy with struct and method using same identifier. That might be nice to fix, but i think it is more important to maintain C++ compatibility in general.

```make
$ LANG=C make
g++ -c -O2 -flto -ftree-vectorize -ftree-slp-vectorize -fvisibility=hidden -D_FORTIFY_SOURCE=2 -DMG_ENABLE_FS=0 -fstack-protector-strong -march=silvermont -mtune=generic -Wall -Wextra -Werror -pedantic -Werror=date-time -Wformat-security -Wduplicated-cond -Wfloat-equal -Wshadow -Wlogical-not-parentheses -Wnull-dereference -D_POSIX_C_SOURCE=200809L -D_XOPEN_SOURCE=700 -D_DEFAULT_SOURCE=1 -D_GNU_SOURCE -std=c++11 -I/usr/include/json-c  -pthread -MD -o main.o main.cpp
In file included from main.cpp:17:
libs/mongoose.h:450:35: error: 'mg_str mg_str(const char*)' hides constructor for 'struct mg_str' [-Werror=shadow]
  450 | struct mg_str mg_str(const char *s);
      |                                   ^
main.cpp: In function 'int main(int, char**)':
main.cpp:512:25: error: invalid conversion from 'void (*)(const void*, int, void*)' to 'void (*)(const void*, size_t, void*)' {aka 'void (*)(const void*, long unsigned int, void*)'} [-fpermissive]
  512 |     mg_log_set_callback(mg_log_syslog, NULL);
      |                         ^~~~~~~~~~~~~
      |                         |
      |                         void (*)(const void*, int, void*)
In file included from main.cpp:17:
libs/mongoose.h:475:33: note:   initializing argument 1 of 'void mg_log_set_callback(void (*)(const void*, size_t, void*), void*)'
  475 | void mg_log_set_callback(void (*fn)(const void *, size_t, void *), void *param);
      |                          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

Thanks for looking into this.

Harvie, 2021-06-14:
I've found that Makefile contains C++ build with following, so i guess that might be needed.

`linux++: WARN += -Wno-shadow  # Ignore "hides constructor for 'struct mg_str'"`

struct and method having same name isn't really a problem, but using such option might hide some more serious problems in the code (even problems laying outside of the mongoose code base)

star-ismael-serrano, 2021-07-12:
I posted the solution here: #1305

Harvie, 2021-07-12:
> I posted the solution here: #1305

Are you sure we are talking about the same issue? I don't think that solution proposed in #1305 will fix #1298 as well, but correct me if i am wrong...

cpq, 2021-07-14:
Yeah, the shadow warning is useful, disabling it is not the best idea.
That said, the `mg_str()` function is nice and concise, changing it to something like `mg_mk_str()` looks ugly.
Maybe there is a nicer solution exist to make a C++ compiler happy.
Anyway, I don't perceive this issue as a Mongoose bug.

closed 2021-07-14 by cpq

Harvie, 2021-07-14:
I wonder what would happen if we had macro like this: `#define mg_str(...) mg_mk_str(__VA_ARGS__)`
That might fix the issue, since there will be no actual shadowing in preprocessed code.

cpq, 2021-07-15:
What if `mg_str` is defined as a static inline function in the header

```c
static inline struct mg_str mg_str(...) { ... }
```

The rationale is that that the static inline could be treated as macro, expanded before the shadow check is performed?

cpq, 2021-07-15:
Macro idea is a nice one.

cpq, 2021-07-15:
Here we go.
Thank you @Harvie

Harvie, 2021-07-15:
Just tested, compiles without warnings. Thank you.

patlecat, 2021-07-16:
The lib is notorious for having problems when compiling in a C++ Windows project, also with VisualStudio. The company has expressed no interest in getting this tackled.

Harvie, 2021-07-16:
@patlecat i think you might be bit biased. 95% of times i had some issues with some of the cesanta's opensource products they were rather helpful and after bit of communication the issue got resolved. except for some feature requests, in some cases i had to add the new feature myself and send it in for review, but it still eventualy got through. which is completely ok, because they are not obliged to implement every feature i propose. Overall i reccomend mongoose products to friends even when i am not in any way affiliated with cesanta. (even though i am working for company which buys mongoose os licences for products we developed, which kinda makes me a customer of cesanta)

Also have you considered using GCC on windows? It should work pretty much the same way as GCC on linux. You can probably use GCC integration available as VisualStudio Code plugin. Perhaps you should point out particular issue rather than genericaly stating that mongoose has issues on windows.

cpq, 2021-07-16:
@patlecat 
Note that mongoose's continuous integration compiles and runs a unit test on many platforms and compilers, including ancient MSVC98 compiler (see https://github.com/cesanta/mongoose/blob/77fae5c562b831c4f92038d3691264ec6c38a237/Makefile#L74-L76) and more modern MCVC 2017 (see https://github.com/cesanta/mongoose/blob/77fae5c562b831c4f92038d3691264ec6c38a237/Makefile#L79-L81). Which are of course Microsoft compilers.

@patlecat said all that, could you be more specific please?

patlecat, 2021-07-16:
> 
> 
> @patlecat i think you might be bit biased. 95% of times i had some issues with some of the cesanta's opensource products they were rather helpful and after bit of communication the issue got resolved. except for some feature requests, in some cases i had to add the new feature myself and send it in for review, but it still eventualy got through. which is completely ok, because they are not obliged to implement every feature i propose. Overall i reccomend mongoose products to friends even when i am not in any way affiliated with cesanta. (even though i am working for company which buys mongoose os licences for products we developed, which kinda makes me a customer of cesanta)
> 
> Also have you considered using GCC on windows? It should work pretty much the same way as GCC on linux (except for OS API will be different). You can probably use GCC integration available as VisualStudio Code plugin. Perhaps you should point out particular issue rather than genericaly stating that mongoose has issues on windows.

Bad experiences can make a person biased but it's still the truth.

commit 77fae5c5 (2021-07-15): Fix #1298 - shadowing struct mg_str constructor