cesanta/mongoose #1305: Unable to compile using Visual Studio

ajtruckle, 2021-07-04:
I created a empty c++ project and added your mongoose header / source files.

I used your sample:

#include "mongoose.h"

```
static const char* s_web_root_dir = ".";
static const char* s_listening_address = "http://localhost:1000";

static void cb(struct mg_connection* c, int ev, void* ev_data, void* fn_data) {
	struct mg_http_serve_opts opts;
	opts.root_dir = s_web_root_dir;

	if (ev == MG_EV_HTTP_MSG) mg_http_serve_dir(c, (struct mg_http_message *) ev_data, &opts);
}

int main(int argc, char* argv[]) {
	struct mg_mgr mgr;
	mg_mgr_init(&mgr);

	mg_http_listen(&mgr, s_listening_address, cb, &mgr);

	for (;;) mg_mgr_poll(&mgr, 1000);

	mg_mgr_free(&mgr);
	return 0;
}
```

But when I try to compile it:

```
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_poll(struct mg_mgr *,int)" (?mg_mgr_poll@@YAXPEAUmg_mgr@@H@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_poll
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_init(struct mg_mgr *)" (?mg_mgr_init@@YAXPEAUmg_mgr@@@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_init
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_free(struct mg_mgr *)" (?mg_mgr_free@@YAXPEAUmg_mgr@@@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_free
1>main.obj : error LNK2019: unresolved external symbol "struct mg_connection * __cdecl mg_http_listen(struct mg_mgr *,char const *,void (__cdecl*)(struct mg_connection *,int,void *,void *),void *)" (?mg_http_listen@@YAPEAUmg_connection@@PEAUmg_mgr@@PEBDP6AXPEAU1@HPEAX3@Z3@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_http_listen
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_http_serve_dir(struct mg_connection *,struct mg_http_message *,struct mg_http_serve_opts *)" (?mg_http_serve_dir@@YAXPEAUmg_connection@@PEAUmg_http_message@@PEAUmg_http_serve_opts@@@Z) referenced in function "void __cdecl cb(struct mg_connection *,int,void *,void *)" (?cb@@YAXPEAUmg_connection@@HPEAX1@Z)
1>  Hint on symbols that are defined and could potentially match:
1>    mg_http_serve_dir
1>mongoose.obj : error LNK2019: unresolved external symbol __WSAFDIsSet referenced in function mg_iotest
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_accept referenced in function accept_conn
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_bind referenced in function mg_open_listener
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_closesocket referenced in function accept_conn
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_connect referenced in function mg_connect_resolved
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_ioctlsocket referenced in function mg_set_non_blocking_mode
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_getsockopt referenced in function connect_conn
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_listen referenced in function mg_open_listener
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_recv referenced in function mg_sock_recv
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_recvfrom referenced in function mg_sock_recv
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_select referenced in function mg_iotest
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_send referenced in function mg_sock_send
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_sendto referenced in function mg_sock_send
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_setsockopt referenced in function mg_open_listener
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_socket referenced in function mg_connect_resolved
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_WSAStartup referenced in function mg_mgr_init
1>mongoose.obj : error LNK2019: unresolved external symbol __imp_WSAGetLastError referenced in function accept_conn
1>C:\Users\ajtru\Source\Repos\MSA WebServer\x64\Debug\MSA WebServer.exe : fatal error LNK1120: 22 unresolved externals

```

ajtruckle, 2021-07-04:
I added this library to my project settings:

Ws2_32.lib

Now the errors are reduced to:

```
Build started...
1>------ Build started: Project: MSA WebServer, Configuration: Debug x64 ------
1>main.cpp
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_poll(struct mg_mgr *,int)" (?mg_mgr_poll@@YAXPEAUmg_mgr@@H@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_poll
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_init(struct mg_mgr *)" (?mg_mgr_init@@YAXPEAUmg_mgr@@@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_init
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_mgr_free(struct mg_mgr *)" (?mg_mgr_free@@YAXPEAUmg_mgr@@@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_mgr_free
1>main.obj : error LNK2019: unresolved external symbol "struct mg_connection * __cdecl mg_http_listen(struct mg_mgr *,char const *,void (__cdecl*)(struct mg_connection *,int,void *,void *),void *)" (?mg_http_listen@@YAPEAUmg_connection@@PEAUmg_mgr@@PEBDP6AXPEAU1@HPEAX3@Z3@Z) referenced in function main
1>  Hint on symbols that are defined and could potentially match:
1>    mg_http_listen
1>main.obj : error LNK2019: unresolved external symbol "void __cdecl mg_http_serve_dir(struct mg_connection *,struct mg_http_message *,struct mg_http_serve_opts *)" (?mg_http_serve_dir@@YAXPEAUmg_connection@@PEAUmg_http_message@@PEAUmg_http_serve_opts@@@Z) referenced in function "void __cdecl cb(struct mg_connection *,int,void *,void *)" (?cb@@YAXPEAUmg_connection@@HPEAX1@Z)
1>  Hint on symbols that are defined and could potentially match:
1>    mg_http_serve_dir
1>C:\Users\ajtru\Source\Repos\MSA WebServer\x64\Debug\MSA WebServer.exe : fatal error LNK1120: 5 unresolved externals
1>Done building project "MSA WebServer.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

```

cpq, 2021-07-09:
Please compile mongoose.c as C, not C++.

star-ismael-serrano, 2021-07-09:
When including mongoose in a c++ project do it in this way:
`extern "C"
{
#include "mongoose.h"
}`

Anyways I thing that should be handled inside the mongoose header adding at the beginning  of it

`#ifdef __cplusplus
extern "C" {
#endif`

And this at the end

`#ifdef __cplusplus
}
#endif`

So the users that use it with c++ projects don't get any problem like this.

ajtruckle, 2021-07-09:
Thanks for the explanation and I agree it would be a useful improvement to the library.

Harvie, 2021-07-12:
> Anyways I thing that should be handled inside the mongoose header adding at the beginning of it

I second that. This is very common solution and i see it almost everywhere when C and C++ gets mixed.

ajtruckle, 2021-07-12:
And I had to add the link to the library. Which I don’t think was documented??

commit 01558695 (2021-07-13): Fix #1305 - add #ifdef __cplusplus for C++ linkage
closed 2021-07-13 by cpq