cesanta/mongoose #1307: Crash on mg_tls_init

djpiper28, 2021-07-04:
- My goal is: running a bot
- My actions were: api server for the bot accepts a connection with mbedtls for the crypto
- My expectation was: no crash
- The result I saw: a crash
- My question is: why crash and how fix?

# Environment

- mongoose version: latest commit as of 04/07/2021 (dd-mm-yyyy)
- custom changes?: n
  - If yes, provide the diff below, and the releavant commit:
- build environment:
  - OS: debian linux
  - Compiler/IDE: gcc/g++
- relevant example (see https://github.com/cesanta/mongoose/tree/master/examples) is: _?_
- relevant documentation topic (see https://cesanta.com/docs/) is: _?_
## MBEDTLS VERSION:

```bash
root@tricebot:~/bin# apt list | grep mbedtls

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libmbedtls-dev/stable,now 2.16.0-1 amd64 [installed]
libmbedtls-doc/stable 2.16.0-1 all
libmbedtls12/stable,now 2.16.0-1 amd64 [installed,automatic]
root@tricebot:~/bin# 
```
(Source code if that helps: https://github.com/djpiper28/CockatriceTournamentBot)
Bind address: `https://127.0.0.1:8000`, SSL Certificates generated by certbot and don't expire for about a month.
Request sent:
```
authtoken=[REDACTED]
gamename=test-deck-hashes/Match3
password=[REDACTED]
playerCount=1
spectatorsAllowed=0
spectatorsNeedPassword=0
spectatorsCanChat=0
spectatorsCanSeeHands=0
onlyRegistered=0
playerDeckVerification=1
playerName=*
deckHash=ie03v62v

```

# COREDUMP:

```gdb
Core was generated by `./botExec'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7fb614bd9700 (LWP 16315))]
(gdb) bt full
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 94430847706812, 11, 94430847706812, 140420008742048, 94430847706817, 0, 0, 0, 0, 0, 0, 140420008741552, 140420008741576, 140420008742083, 140420008741552}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007fb615628535 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x0, sa_sigaction = 0x0}, sa_mask = {__val = {0 <repeats 11 times>, 17179869184, 140419795065520, 0, 140420008740912, 140420008741168}}, sa_flags = 347965488, sa_restorer = 0x1000}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007fb61567f508 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fb61578a28d "%s\n") at ../sysdeps/posix/libc_fatal.c:181
        ap = {{gp_offset = 24, fp_offset = 0, overflow_arg_area = 0x7fb614bd8940, reg_save_area = 0x7fb614bd88d0}}
        fd = 2
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
        written = <optimized out>
#3  0x00007fb615685c1a in malloc_printerr (str=str@entry=0x7fb615788376 "corrupted size vs. prev_size") at malloc.c:5341
No locals.
#4  0x00007fb615685e74 in malloc_consolidate (av=av@entry=0x7fb608000020) at malloc.c:4488
        fb = <optimized out>
        maxfb = 0x7fb608000078
        p = <optimized out>
        nextp = 0x0
        unsorted_bin = 0x7fb608000080
        first_unsorted = <optimized out>
        nextchunk = <optimized out>
        size = <optimized out>
        nextsize = <optimized out>
        prevsize = <optimized out>
        nextinuse = <optimized out>
        bck = <optimized out>
        fwd = <optimized out>
#5  0x00007fb615688a58 in _int_malloc (av=av@entry=0x7fb608000020, bytes=bytes@entry=1960) at malloc.c:3695
        nb = 1968
        idx = 78
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = <optimized out>
        tcache_nb = <optimized out>
        tc_idx = <optimized out>
        return_cached = <optimized out>
        __PRETTY_FUNCTION__ = "_int_malloc"
#6  0x00007fb61568b1a2 in __libc_calloc (n=<optimized out>, elem_size=<optimized out>) at malloc.c:3428
        av = <optimized out>
        oldtop = 0x7fb608016240
--Type <RET> for more, q to quit, c to continue without paging--c
        p = <optimized out>
        bytes = 1960
        sz = 1960
        csz = <optimized out>
        oldtopsize = 44480
        mem = <optimized out>
        clearsize = <optimized out>
        nclears = <optimized out>
        d = <optimized out>
        hook = <optimized out>
        __PRETTY_FUNCTION__ = "__libc_calloc"
#7  0x000055e26520394d in mg_tls_init (c=0x7fb608011ab0, opts=0x7ffc8070c950) at src/tls.c:68
        tls = 0x55e265204236 <mg_url_is_ssl(char const*)+64>
        rc = 0
        ca = 0x0
        cert = 0x7fb614bd8df7 ""
        certkey = 0x55e2653c0770 "https://127.0.0.1:8000"
        __func__ = "mg_tls_init"
#8  0x000055e265117817 in eventHandler (c=0x7fb608011ab0, event=4, ev_data=0x0, fn_data=0x7ffc8070c920) at /root/Tricebot/CockatriceTournamentBot/src/api_server.c:677
        s = 0x7fb608003cf0
        api = 0x7ffc8070c920
#9  0x000055e2651f9a1b in mg_call (c=0x7fb608011ab0, ev=4, ev_data=0x0) at src/event.c:13
No locals.
#10 0x000055e2652025b3 in accept_conn (mgr=0x7fb614bd8e90, lsn=0x7fb608000b20) at src/sock.c:486
        buf = "127.0.0.1:55672\000a\340 e\342U\000\000\000\000\000\000\000\000\000\000a\340 e\342U\000"
        c = 0x7fb608011ab0
        usa = {sa = {sa_family = 2, sa_data = "\331x\177\000\000\001\000\000\000\000\000\000\000"}, sin = {sin_family = 2, sin_port = 30937, sin_addr = {s_addr = 16777343}, sin_zero = "\000\000\000\000\000\000\000"}, sin6 = {sin6_family = 2, sin6_port = 30937, sin6_flowinfo = 16777343, sin6_addr = {__in6_u = {__u6_addr8 = "\000\000\000\000\000\000\000\000\060\216\275\024\266\177\000", __u6_addr16 = {0, 0, 0, 0, 36400, 5309, 32694, 0}, __u6_addr32 = {0, 0, 347967024, 32694}}}, sin6_scope_id = 347967064}}
        sa_len = 16
        fd = 6
        __func__ = "accept_conn"
#11 0x000055e265202f00 in mg_mgr_poll (mgr=0x7fb614bd8e90, ms=100) at src/sock.c:675
        c = 0x7fb608000b20
        tmp = 0x0
        now = 1625423402179
        __func__ = "mg_mgr_poll"
#12 0x000055e2651185ed in pollingThread (apiIn=0x7ffc8070c920) at /root/Tricebot/CockatriceTournamentBot/src/api_server.c:1064
        mgr = {conns = 0x7fb608011ab0, dns4 = {url = 0x55e265222711 "udp://8.8.8.8:53", c = 0x0}, dns6 = {url = 0x55e265222728 "udp://[2001:4860:4860::8888]:53", c = 0x0}, dnstimeout = 3000, nextid = 3, userdata = 0x0}
        c = 0x7fb608000b20
        api = 0x7ffc8070c920
        cont = 1
#13 0x00007fb615af1fa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
        ret = <optimized out>
        pd = <optimized out>
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140420008744704, -3554313377566241517, 140722463360270, 140722463360271, 140420008744704, 0, 3585045260700405011, 3585042937650400531}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#14 0x00007fb6156ff4cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

cpq, 2021-07-14:
Could you show a snippet that prepares the opts struct and calls init please?

djpiper28, 2021-07-14:
```c
    if (event == MG_EV_ACCEPT) {
        //Init connection struct
        struct ServerConnection *s = (struct ServerConnection *)
                                     malloc(sizeof(struct ServerConnection));
        initServerConnection(s, api);
        
        c->fn_data = (void *) s;
        
        if (mg_url_is_ssl(api->config.bindAddr)) {
            mg_tls_init(c, &api->opts);
        }
    }
    ```
    under the function that gets polled.

cpq, 2021-07-16:
Looking at the backtrace, this is suspicious:

```
#7  0x000055e26520394d in mg_tls_init (c=0x7fb608011ab0, opts=0x7ffc8070c950) at src/tls.c:68
        tls = 0x55e265204236 <mg_url_is_ssl(char const*)+64>
        rc = 0
        ca = 0x0
        cert = 0x7fb614bd8df7 ""
        certkey = 0x55e2653c0770 "https://127.0.0.1:8000"
```

The `cert` should be a valid string pointing to a file, so as `certkey`. In your case, they appear garbled.
Could you break in `mg_tls_init` and share the gdb's `p *opts` please?

djpiper28, 2021-07-16:
I have lost my coredump but I can grab the info of what the opts struct would be and the data it was initialised with.

```c
void tb_initServer(struct tb_apiServer *server,
                   struct triceBot *triceBot,
                   struct Config config) {
    server->bottleneck = PTHREAD_MUTEX_INITIALIZER;
    server->config = config;
    server->triceBot = triceBot;
    server->running = 0;
// vvvvvvv
    if (mg_url_is_ssl(server->config.bindAddr)) {
        server->opts.cert = server->config.cert;
        server->opts.certkey = server->config.certkey;
        server->opts.ca = NULL;//server->config.cert;
    }

// ^^^^^^
    
    server->replayFolerWildcard = (char *) malloc(sizeof(char) * BUFFER_LENGTH);
    snprintf(server->replayFolerWildcard,
             BUFFER_LENGTH,
             "/%s/**/*", server->config.replayFolder);
}
```

So the opts are read from my config file which has the values of:
```c
cert=fullchain.pem
certkey=privkey.pem
```
Both of these files do exist and are copied from my main ssh key storage folder.

djpiper28, 2021-07-16:
as the coredump is from 12 days ago it appears that i forgot to copy it and my deamons cleaned it up. I am assuming that the memory was garbled at some point however I have no evidence of this

cpq, 2021-07-16:
Thank you! How about other attributes of opts? Are they NULL or contain garbage (which could happen if server pointer was malloc-ed but not zeroed)?

djpiper28, 2021-07-16:
server is not zeroed as it is a heap variable so would indeed start with noise in memory.

djpiper28, 2021-07-16:
I have have added the recommended fix however as the issue only appears randomly I do not know if it works.

cpq, 2021-07-16:
Well, one way is to fill server->opts from /dev/urandom instead of memset with 0, and see what happens.

djpiper28, 2021-07-16:
I will try that

djpiper28, 2021-07-16:
I tried randomising the memory with
```c

#include <stdio.h>
#include <sys/time.h>
#include <unistd.h>
(...)
struct timeval seedTime;
gettimeofday(&seedTime, NULL);
srand(seedTime.tv_usec);
for (int i = 0; i < sizeof(server->opts); i++) {
    ((char *) &server->opts)[i] = (char) rand();
}
```
instead of memset and the crash does not occur

cpq, 2021-07-20:
@djpiper28 thank you!

So let's summarise - did zeroing the structure before calling `mg_tls_init()` help?
Can we mark this issue as resolved?

djpiper28, 2021-07-20:
No.

cpq, 2021-07-21:
Ok. Then, could you provide a backtrace for the case when TLS opts are correctly initialised: zeroed, then some fields are set.
Please use the latest source.

djpiper28, 2021-07-21:
i cannot get the program to crash reliably. it crashes randomly every few days and I do not know why. when it crashes next i will send the backtrace your way.

cpq, 2021-07-21:
@djpiper28 great, thank you.

Please consider using sanitizer, if you don't use it already. Example CFLAGS:

```make
CFLAGS +=  -fsanitize=address,undefined
CFLAGS += -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion
CFLAGS += -g3 -O3
```

djpiper28, 2021-07-24:
I cannot get the `-fsanitize=address,undefined` cflag to play nice with `pthread_mutex_t` as it just deadlocks for no good reason. I will investigate with valgrind as that works well.

djpiper28, 2021-07-24:
valgrind only shows this error
```
==32705== Conditional jump or move depends on uninitialised value(s)
==32705==    at 0x12F593: botEventHandler(mg_connection*, int, void*, void*) (bot.c:1395)
==32705==    by 0x20A589: mg_call(mg_connection*, int, void*) (event.c:9)
==32705==    by 0x214240: mg_mgr_poll(mg_mgr*, int) (sock.c:560)
==32705==    by 0x12FA8A: botThread(void*) (bot.c:1498)
==32705==    by 0x4ED4FA2: start_thread (pthread_create.c:486)
==32705==    by 0x53084CE: clone (clone.S:95)
```
but that can be ignored I think

djpiper28, 2021-07-24:
as that is from my code so idk where spooky memory comes from

djpiper28, 2021-07-24:
I've attacked my program with a lot of debuggers and stuff and I cannot get this crash to happen again. I'm going to call this bug memory corruption by cosmic rays for my sanity. If this bug reappears I'll grab that coredump and disect it. Until then I will close this issue and wonder what happened to my system.

closed 2021-07-24 by djpiper28