cesanta/mongoose #1308: clang compile error too few arguments of mbedtls_pk_parse_key

Tebeye, 2021-07-06:
I want to compile the mongoose in my ubuntu but following error occurs, I've installed mbedtls for three times and check older versions. But the error still remains.
```
~/mongoose$ clang  mongoose.c test/unit_test.c -O3 -g3 -W -Wall -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -DMG_MAX_HTTP_HEADERS=5 -DMG_ENABLE_LINES -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1   -DMG_ENABLE_MBEDTLS=1 -I//include -I/usr/include -DMG_ENABLE_IPV6=1 -fsanitize=address -coverage -L//lib -lmbedtls -lmbedcrypto -lmbedx509 -g -o unit_test
src/tls.c:145:61: error: too few arguments to function call, expected 7, have 5
                                    strlen(key) + 1, NULL, 0)
                                                            ^
/usr/local/include/mbedtls/pk.h:680:1: note: 'mbedtls_pk_parse_key' declared
      here
int mbedtls_pk_parse_key( mbedtls_pk_context *ctx,
^
src/tls.c:146:60: error: too few arguments to function call, expected 5, have 3
             : mbedtls_pk_parse_keyfile(&tls->pk, key, NULL);
               ~~~~~~~~~~~~~~~~~~~~~~~~                    ^
/usr/local/include/mbedtls/pk.h:734:1: note: 'mbedtls_pk_parse_keyfile' declared
      here
int mbedtls_pk_parse_keyfile( mbedtls_pk_context *ctx,
^
2 errors generated.
biss@biss:~/mongoose$ clang  mongoose.c test/unit_test.c -O3 -g3 -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -DMG_MAX_HTTP_HEADERS=5 -DMG_ENABLE_LINES -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1   -DMG_ENABLE_MBEDTLS=1 -I//include -I/usr/include -DMG_ENABLE_IPV6=1 -fsanitize=address -coverage -L//lib -lmbedtls -lmbedcrypto -lmbedx509 -g -o unit_test
src/tls.c:145:61: error: too few arguments to function call, expected 7, have 5
                                    strlen(key) + 1, NULL, 0)
                                                            ^
/usr/local/include/mbedtls/pk.h:680:1: note: 'mbedtls_pk_parse_key' declared
      here
int mbedtls_pk_parse_key( mbedtls_pk_context *ctx,
^
src/tls.c:146:60: error: too few arguments to function call, expected 5, have 3
             : mbedtls_pk_parse_keyfile(&tls->pk, key, NULL);
               ~~~~~~~~~~~~~~~~~~~~~~~~                    ^
/usr/local/include/mbedtls/pk.h:734:1: note: 'mbedtls_pk_parse_keyfile' declared
      here
int mbedtls_pk_parse_keyfile( mbedtls_pk_context *ctx,
^
2 errors generated.
```

# OS Info
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

# Environment
- build environment:
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

cpq, 2021-07-13:
Apparently, mbedtls changed that API recently.
Let us use a conditional to adopt to the recent version.

cpq, 2021-07-13:
Pushed https://github.com/cesanta/mongoose/commit/d9680df510a4f85cd411fe6f9c8094bb7b8f0084, please update to the latest version and retry your build.

closed 2021-07-13 by cpq