cesanta/mongoose #1309: mg_tls_init crashes on MG_EV_ACCEPT event using OpenSSL

nejatafshar, 2021-07-08:
- My goal is: To have a simple https server like the one in the examples: `http-restful-server`
- My actions were: Using the code in `http-restful-server` with https url and linking my app to OpenSSL and using the ssl certificates in  `http-restful-server` example.
- My expectation was: initializing ssl on `MG_EV_ACCEPT` event
- The result I saw: crash on `mg_tls_init`
- My question is: What is the possible cause of crashing in `mg_tls_init`?

I should note that if I switch to version 6.18 and change the code to be compatible with that version, every thing works as expected.
By declaring `mg_tls_opts` variable outside the handler function for example  as a class member sometimes does not crash but gives an error event in handler with this error string: `parse(PJPÍP): err 5`. But also crashes randomly in different runs.

# Environment

- mongoose versions: latest commit (Jul 2, 2021), 7.2, 7.1
- build environment:
  - OS: Windows 10 x64
  - Compiler/IDE: VS 2019 (19.26.28806)
  - OpenSSL 1.1.1k

cpq, 2021-07-13:
http-restful-server example has been updated to use both HTTP and HTTPS ports.
Please update and report whether you still see an issue.

cpq, 2021-07-20:
No response, closing.
Feel free to reopen if required.

closed 2021-07-20 by cpq

ayouboss, 2021-12-23:
I am still seeing the issue unfortunately.