cesanta/mongoose #1312: seg fault on mg_http_serve_dir

Samega7Cattac, 2021-07-12:
- My goal is: Create a HTTPS webserver + websockets and other stuff
- My actions were: By setting the `ms` on mg_mgr_poll to something higher than 50 (100 for example) it made my poll function crash on `mg_http_serve_dir`, probably my poll function is too big and 50ms wasn't enough, that's what made me increase the value and found this crash, my previous issue was not sending index.html (but no crash, output seems normal)
- My expectation was: Serve my index.html
- The result I saw: seg fault on:
```
msvcrt.dll!ungetwc (Unknown Source:0)
msvcrt.dll!msvcrt!_vsnprintf_l (Unknown Source:0)
msvcrt.dll!vsnprintf (Unknown Source:0)
mg_vasprintf(char ** buf, size_t size, const char * fmt, va_list ap) (mongoose.c:4100)
mg_vprintf(struct mg_connection * c, const char * fmt, va_list ap) (mongoose.c:2128)
mg_printf(struct mg_connection * c, const char * fmt) (mongoose.c:2138)
mg_http_serve_file(struct mg_connection * c, struct mg_http_message * hm, const char * path, const char * mime, const char * hdrs) (mongoose.c:964)
mg_http_serve_dir(struct mg_connection * c, struct mg_http_message * hm, struct mg_http_serve_opts * opts) (mongoose.c:1287)
WebServerService::OnEvent(mg_connection * connection, int event, void * event_args, void * function_args)
.
.
.
```
- My question is: Why? and how to solve it?

# Environment

- mongoose version: 7.2
- custom changes?: commits from master until f46453f80f972e28dd8d7ba2f7de9499020b3dbe
- build environment:
  - OS: Windows 10 (VM) + Msys2
  - Compiler/IDE: Mingw-gcc 10.3 / Visual Studio Code
- relevant example (see https://github.com/cesanta/mongoose/tree/master/examples) is: _?_
- relevant documentation topic (see https://cesanta.com/docs/) is: _?_

Ask for code snippets if needed

Samega7Cattac, 2021-07-13:
on line `4100` in the cpp `vsnprintf` have `ap_copy` empty (same for `ap`).

cpq, 2021-07-13:
Could you rebuild with `-g3` please ? It would be good to see a verbose backtrace.

cpq, 2021-07-14:
This line strikes attention:
```
    hdrs=0x82040086 <error: Cannot access memory at address 0x82040086>)
```

Could you show a snippet which calls `mg_http_serve_dir()` please? Maybe the opts structure has garbage?

Samega7Cattac, 2021-07-14:
```
void
WebServerService::OnEvent(struct mg_connection *connection, int event,
                            void *event_args, void *function_args)
{
    // Other stuff
    
    switch(event)
    {
        // If it's accepting a connection
    case MG_EV_ACCEPT:
        {
            // If the ulr is asking for ssl (https)
            if (mg_url_is_ssl(WEBSERVICE_LISTENING_ADDRESS))
            {
                // Creates the TLS struct and sets the certificate and his key
                struct mg_tls_opts tls_opts = {
                    .cert = path_cert.c_str(), // the vars are not in this code snippet but they are correct
                    .certkey = path_key.c_str()
                };

                // Initializes TLS for this connection
                mg_tls_init(connection, &tls_opts);
            }
            break;
        }
        // If it's a HTTP request
    case MG_EV_HTTP_MSG:
        {
            // Converts the event args pointer to a struct http_messsage pointer
            struct mg_http_message *http_message
                = (struct mg_http_message *)event_args;

            // Gets the user that is requesting
            User * user = GetUser(http_message);

            // If the user is valid and it's trying to log in
            if (user != nullptr && mg_http_match_uri(http_message, "/login"))
            {
                // Send back the username and token for the user
                mg_printf(connection,
                            "HTTP/1.1 200 OK\r\nTransfer-Encoding: chunked\r\n\r\n");
                mg_http_printf_chunk(connection,
                                        "{\"user\":\"%s\",\"token\":\"%s\"}\n",
                                        user->GetUsername().c_str(),
                                        user->GetToken().c_str());
                mg_http_printf_chunk(connection, "");
            }
            // If the user is valid and tries to access the modules api endpoint
            if (user != nullptr && mg_http_match_uri(http_message, "/module_api"))
            {
                // Switches the connection to Websockets
                // (from now on it will catch websockets events from this connection)
                mg_ws_upgrade(connection, http_message, NULL);
            }
            else
            {
                // Path to the www folder
                std::string path_www =
                    service_vars->GetInstallationPath() + WEBSERVICE_WEB_ROOT_PATH;

                // Gets resource from the www dir (it contains an index.html, the startup)
                // This should be the default
                struct mg_http_serve_opts opts;
                opts.root_dir = path_www.c_str();
                mg_http_serve_dir(connection, http_message, &opts);
            }
            break;
        }

        // If it receives a Websocket package
    case MG_EV_WS_MSG:
        {
            // Converts the event data pointer to the websocket data
            struct mg_ws_message *websocket_message =
                                    (struct mg_ws_message *)event_args;

            if (strcmp(websocket_message->data.ptr, "get_analysis_points") == 0)
            {

                std::vector<std::string> analysis_points = {
                    "[Lucy,",
                    "Jumbo XS,",
                    "Jumbo 350,"
                };

                for (int i = 1; i <= 15; ++i)
                {
                    std::string point = "R";
                    point.append(itoa(i, nullptr, 10));
                    point.append(",");
                    analysis_points.push_back(point);
                }

                for (int i = 17; i <= 20; ++i)
                {
                    std::string point = "R";
                    point.append(itoa(i, nullptr, 10));
                    point.append(",");
                    analysis_points.push_back(point);
                }

                for (int i = 23; i <= 73; ++i)
                {
                    std::string point = "R";
                    point.append(itoa(i, nullptr, 10));
                    point.append(",");
                    analysis_points.push_back(point);
                }

                // Sends the data back to the sender
                mg_ws_send(connection, analysis_points.data()->c_str(),
                            analysis_points.data()->length(), WEBSOCKET_OP_TEXT);
            }
            // Delets the connection data
            mg_iobuf_delete(&connection->recv, connection->recv.len);
            break;
        }
    }
}
```

Samega7Cattac, 2021-07-14:
True, setting the other fields to NULL solved the issue.

closed 2021-07-14 by Samega7Cattac