cesanta/mongoose #1315: Not all C compiler does supports `#pragma once`

edubart, 2021-07-18:
I am using a C compiler that does not support `#pragma once`, then when including `mongoose.h` twice I get compile errors due to double declarations. Please add ifdef guards in `mongoose.h`, that is:

```c
#ifndef MONGOOSE_H
#define MONGOOSE_H

/* header code */

#endif
```

cpq, 2021-07-18:
@edubart thanks. Could you elaborate what is the toolchain you're using, the target platform and environment, please?

edubart, 2021-07-18:
> @edubart thanks. Could you elaborate what is the toolchain you're using, the target platform and environment, please?

I'm using https://github.com/vnmakarov/mir for jiting C code.

edubart, 2021-07-18:
I've just made the PR #1317 with a possible fix for this.

commit 3982e065 (2021-07-19): Fix #1315, fix #1317 - use #ifndef instead of #pragma once for compilers that do not support #pragma once
closed 2021-07-19 by cpq