cesanta/mongoose #1316: Mongoose cannot serve html files encoded as UTF8 with BOM

Jaifroid, 2021-07-18:
- My goal is: Serve web pages on my project via localhost
- My actions were: I shared the directory with mongoose
- My expectation was: HTML pages would be served as web pages
- The result I saw: index.html was served as a text file instead of HTML
- My question is: I believe this is because my files are encoded as UTF8 with BOM. The same pages can be served successfully with httpd/Apache. Please note that I do not have the option to remove the BOM, because the project requires all files to be encoded as UTF8 with BOM.

# Environment

- mongoose version: _7.3.2_
- custom changes?: _n_
  - If yes, provide the diff below, and the releavant commit:
- build environment:
  - OS: _Windows 11_
  - Compiler/IDE: _N/A_
- relevant example (see https://github.com/cesanta/mongoose/tree/master/examples) is: _See screenshot_
- relevant documentation topic (see https://cesanta.com/docs/) is: _N/A_

This is how my index.html is served by mongoose:

[image]

cpq, 2021-07-19:
Can't reproduce that - files with BOM are served just fine..

Could you use curl utility or some other tool like Postman to check an exact binary response from Mongoose, including headers, and share it here please?

Jaifroid, 2021-07-19:
OK will do.

Jaifroid, 2021-07-19:
Hi, below is the output of `Invoke-WebRequest` (PowerShell equivalent of curl). As you can see, it looks normal. So I checked the page in Chromium, and again it produces a text representation of the HTML. However, trying the same in Firefox, I get the page served properly. So it's possible the issue is with Chromium. On the other hand, when I spin up an Apache server in Windows Subsystem for Linux (Ubuntu running "beneath" Windows), the same Chromium install on Windows can access the page properly. I had this both on Windows 10 (just now) and previously on Windows 11 (which is the reason I was trying out mongoose -- I needed a quick web server on a fresh install of Windows 11).

```
PS C:\mongoose> Invoke-WebRequest -uri http://localhost:8000

StatusCode        : 200
StatusDescription : OK
Content           : {60, 33, 100, 111…}
RawContent        : HTTP/1.1 200 OK
                    Content-Length: 164

                    <!doctype html>
                    <html>
                    <head>
                        <meta http-equiv="refresh" content="0; url=www/index.html">
                    <title>Redirection to index.html</title>
                    <body>
                    </body>
                    </h…
Headers           : {[Content-Length, System.String[]]}
RawContentLength  : 164
RelationLink      : {}

```

Jaifroid, 2021-07-19:
Ah I just realized that was the output of the redirect. Below is the output of the file it is redirected to. This gives more of a clue: it looks like mongoose is indeed serving the BOM, whereas it should probably filter it out. I guess that's what the ??? is.

```
PS C:\mongoose> Invoke-WebRequest -uri http://localhost:8000/www/index.html

StatusCode        : 200
StatusDescription : OK
Content           : {239, 187, 191, 60…}
RawContent        : HTTP/1.1 200 OK
                    Content-Length: 77735

                    ???<!DOCTYPE html>
                    <html>
                    <head>
                        <meta charset="utf-8">

                        <title>Kiwix JS PWA</title>
                        <meta name="description" content="Offline Wikipedia reader">
                    …
Headers           : {[Content-Length, System.String[]]}
RawContentLength  : 77735
RelationLink      : {}

```

Jaifroid, 2021-07-19:
Under `Content` above you see `{239, 187, 191, 60…}`. The first three numbers correspond to HEX `0xEF`, `0xBB`, `0xBF`, which is the BOM. I think this confirms my original suspicion. Clearly Apache suppresses the BOM when returning a standard web page. Maybe this could be an option in the mongoose configuration? It would only take a quick regex to search/replace `/^\xEF\xBB\xBF/`.

cpq, 2021-07-19:
I don't see HTTP headers in your outputl. Please use normal tools that show the required information.

Jaifroid, 2021-07-19:
Um, I provided the requested information. The response headers are 

`HTTP/1.1 200 OK
Content-Length: 77735`

I can assure you that `Invoke-WebRequest` is a perfectly "normal" and trustworthy tool on Windows ;-) (my issue relates to Windows). However, if you needed the request headers as well, here is git's curl version, though it's less useful because it doesn't show the BOM which is clearly in the binary content returned (as revealed by `Invoke-WebRequest` above):

```
*   Trying ::1...
* TCP_NODELAY set
*   Trying 127.0.0.1...
* TCP_NODELAY set
* Connected to localhost (127.0.0.1) port 8000 (#0)
> GET /www/index.html HTTP/1.1
> Host: localhost:8000
> User-Agent: curl/7.55.1
> Accept: */*
>> GET /www/index.html HTTP/1.1
> Host: localhost:8000
> User-Agent: curl/7.55.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 77735
<
﻿<!DOCTYPE html>
<html>
<head>
    <meta charset="utf-8">

    <title>Kiwix JS PWA</title>
```

> How do you serve the file?

`mongoose -d D:\Users\xxxxx\Source\Repos\kiwix-js-windows`

> there is no Content-Type sent

Interesting. So the issue is with mongoose after all?

Jaifroid, 2021-07-19:
For reference, I have compared the above output with the output of a file served from a different directory that contains an `index.html` of a related project that does NOT have the BOM. This document is shown correctly in Chromium (it is interpreted as HTML); however, it doesn't have ContentType either, nor ETAG etc. Curl output below, and below that Invoke-WebRequest so you can see there is no BOM. The only difference between the two, which makes Chromium able to interpret the page as HTML, is the absence of the BOM.
```
*   Trying ::1...
* TCP_NODELAY set
*   Trying 127.0.0.1...
* TCP_NODELAY set
* Connected to localhost (127.0.0.1) port 8000 (#0)
> GET /www/index.html HTTP/1.1
> Host: localhost:8000
> User-Agent: curl/7.55.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 46280
<
<!DOCTYPE html>
<html>
    <head>
        <meta charset="utf-8">

        <title>Kiwix</title>
        <meta name="description" content="Offline Wikipedia reader">
        <meta name="viewport" content="width=device-width, initial-scale=1.0">
        <!--
        Kiwix (offline Wikipedia reader) - HTML5/Javascript version
        Home page : https://www.kiwix.org
```
And `Invoke-WebRequest`:
```
PS C:\mongoose> Invoke-WebRequest -uri http://localhost:8000/www/index.html

StatusCode        : 200
StatusDescription : OK
Content           : {60, 33, 68, 79…}
RawContent        : HTTP/1.1 200 OK
                    Content-Length: 46280

                    <!DOCTYPE html>
                    <html>
                        <head>
                            <meta charset="utf-8">

                            <title>Kiwix</title>
                            <meta name="description" content="Offline Wikipedia rea…
Headers           : {[Content-Length, System.String[]]}
RawContentLength  : 46280
RelationLink      : {}
```

cpq, 2021-07-20:
Ah. I see what is going on.
You are using mongoose executable, not the library.
Please confirm.

Jaifroid, 2021-07-20:
> Ah. I see what is going on.
> You are using mongoose executable, not the library.
> Please confirm.

Confirmed! For my use case, that's the attraction of mongoose: one simple exe and I can serve a directory tree with zero (or very little) config.

cpq, 2021-07-20:
I see, thank you. The solution of the problem is this: add `-S #.shtml` argument to the executable when starting, or add exactly the same line to the `mongoose.conf` file.

Explanation:
- Mongoose executable currently does not set `Content-Type` header for the SSI files
- The default SSI pattern is `#.html` which includes all html files
- The `-S ..` overrides default setting

In the mean time, we'll patch the executable to add the `Content-Type` header.

Jaifroid, 2021-07-20:
Thanks, I'll try that command, though just to be clear we're not using *.shtml files (I haven't seen those in years!!). Just plain old *.html, which ought to work out of the box if I've understood you correctly.

cpq, 2021-07-20:
Mongoose executable recognises SSI files by file name pattern, which is set to `#.html` - meaning, html files are treated as SSI. You could add SSI directives into html files and they'll work. This is handy for making websites with common headers/footers.

Jaifroid, 2021-07-21:
Thanks! The latest version works great wihtout needing the -s switch.
Just before downloading the new version, I tested the old one with the -s #.shtml option you suggested, but unfortunately that didn't work (same result as before). Clearly adding the Content-Type did the trick.
Thank you for this -- my issue is now fixed, so I'm closing.

closed 2021-07-21 by Jaifroid

cpq, 2021-07-21:
Thank you.
Please note: you've used `-s` switch, but should have been using `-S` switch (capitalized). That makes a dfference.

Jaifroid, 2021-07-21:
Sorry, that was just my typing here. I did use capital letter. I kept the old binary, so have just tried it again, but it just struck me you probably meant `-S *.shtml` rather than `-S #.shtml`. I confirm that if I use `*.shtml` the page is served correctly by the old mongoose binary. Obviously, though, I'll use the new one. Thanks again.