cesanta/mongoose #1318: make error in ubuntu 16.04

remainfox, 2021-07-19:
make error as follow
os: ubuntu 16.04

```
root@brian:mongoose# 
root@brian:mongoose# make
clang  mongoose.c -O3 -g3 -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -I../mbedtls/include   -DMG_MAX_HTTP_HEADERS=5 -DMG_ENABLE_LINES -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1   -DMG_ENABLE_MBEDTLS=1 -I/usr/include -I/usr/include -c -o /tmp/x.o && nm /tmp/x.o | grep ' T' | grep -v 'mg_' ; test $? = 1
clang  mongoose.c test/unit_test.c -O3 -g3 -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -I../mbedtls/include   -DMG_MAX_HTTP_HEADERS=5 -DMG_ENABLE_LINES -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1   -DMG_ENABLE_MBEDTLS=1 -I/usr/include -I/usr/include -DMG_ENABLE_IPV6=1 -fsanitize=address -coverage -L/usr/lib -L../mbedtls/library -lmbedtls -lmbedcrypto -lmbedx509 -g -o unit_test
ASAN_OPTIONS=  ./unit_test
2021-07-19 06:50:07  I sock.c:474:mg_listen      1 accepting on http://127.0.0.1:12344
2021-07-19 06:50:07    sock.c:369:mg_connect     2 -> http://127.0.0.1:12344
2021-07-19 06:50:07    sock.c:402:accept_conn    3 accepted 127.0.0.1:32936
2021-07-19 06:50:07    sock.c:285:close_conn     2 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     4 -> http://127.0.0.1:12344
2021-07-19 06:50:07    sock.c:229:read_conn      3   00100000000011       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     3 closed
2021-07-19 06:50:07    sock.c:402:accept_conn    5 accepted 127.0.0.1:32938
2021-07-19 06:50:07    sock.c:285:close_conn     4 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     6 -> http://127.0.0.1:12344
2021-07-19 06:50:07    sock.c:229:read_conn      5   00100000000011       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     5 closed
2021-07-19 06:50:07    sock.c:402:accept_conn    7 accepted 127.0.0.1:32940
2021-07-19 06:50:07    sock.c:285:close_conn     6 closed
2021-07-19 06:50:07    sock.c:285:close_conn     7 closed
2021-07-19 06:50:07    sock.c:285:close_conn     1 closed
2021-07-19 06:50:07  I net.c:122:mg_mgr_free     All connections closed
2021-07-19 06:50:07    sock.c:369:mg_connect     1 -> udp://time.google.com:123
2021-07-19 06:50:07    sock.c:369:mg_connect     2 -> udp://8.8.8.8:53
2021-07-19 06:50:07  E sntp.c:53:mg_sntp_send    1 wait until resolved
2021-07-19 06:50:07    dns.c:165:dns_cb          1 time.google.com resolved to 216.239.35.12
2021-07-19 06:50:07    sntp.c:60:mg_sntp_send    0x4 request sent, ct 1626677407, next at 1626681007
2021-07-19 06:50:07    sntp.c:39:sntp_cb         1626677407.3162124933, next at 1626681007
2021-07-19 06:50:07    sock.c:285:close_conn     2 closed
2021-07-19 06:50:07    sock.c:285:close_conn     1 closed
2021-07-19 06:50:07  I net.c:122:mg_mgr_free     All connections closed
2021-07-19 06:50:07  E sntp.c:18:mg_sntp_parse   wrong version
2021-07-19 06:50:07  E sntp.c:16:mg_sntp_parse   corrupt packet
2021-07-19 06:50:07    sock.c:369:mg_connect     1 -> http://google.com
2021-07-19 06:50:07    sock.c:369:mg_connect     2 -> udp://127.0.0.1:12345
2021-07-19 06:50:07  E event.c:18:mg_error       1 DNS timeout
2021-07-19 06:50:07    sock.c:285:close_conn     1 closed
2021-07-19 06:50:07    sock.c:285:close_conn     2 closed
2021-07-19 06:50:07  I net.c:122:mg_mgr_free     All connections closed
2021-07-19 06:50:07  I unit_test.c:939:test_time g_timers: (nil)
2021-07-19 06:50:07  I sock.c:474:mg_listen      1 accepting on http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:369:mg_connect     2 -> http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:402:accept_conn    3 accepted 127.0.0.1:40614
2021-07-19 06:50:07    sock.c:285:close_conn     2 closed
2021-07-19 06:50:07    sock.c:229:read_conn      3   00100000000010       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     3 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     4 -> http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:402:accept_conn    5 accepted 127.0.0.1:40616
2021-07-19 06:50:07  I http.c:496:getrange       bytes=5-10
2021-07-19 06:50:07    sock.c:285:close_conn     4 closed
2021-07-19 06:50:07    sock.c:229:read_conn      5   00100000000010       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     5 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     6 -> http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:402:accept_conn    7 accepted 127.0.0.1:40618
2021-07-19 06:50:07  I http.c:496:getrange       bytes=300-
2021-07-19 06:50:07    sock.c:285:close_conn     6 closed
2021-07-19 06:50:07    sock.c:229:read_conn      7   00100000000010       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     7 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     8 -> http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:402:accept_conn    9 accepted 127.0.0.1:40620
2021-07-19 06:50:07  I http.c:496:getrange       bytes=999-
2021-07-19 06:50:07    sock.c:285:close_conn     8 closed
2021-07-19 06:50:07    sock.c:229:read_conn      9   00100000000010       0 -1/512 err 115
2021-07-19 06:50:07    sock.c:285:close_conn     9 closed
2021-07-19 06:50:07    sock.c:369:mg_connect     10 -> http://127.0.0.1:12349
2021-07-19 06:50:07    sock.c:402:accept_conn    11 accepted 127.0.0.1:40622
[... 419 lines cut ...]
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:13    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
2021-07-19 06:50:14    sock.c:229:read_conn      3   01000100000010   16384 0/512 err 11
FAILURE test/unit_test.c:677: ok == 200
Makefile:60: recipe for target 'test' failed
make: *** [test] Error 1
root@brian:mongoose# 
```

cpq, 2021-07-19:
What if you increase the loop value on line 676 from 500 to a higher value, like 5000 ?

cpq, 2021-07-22:
@remainfox did you manage to repeat the test with the increased count?

cpq, 2021-07-25:
No response, closing this.

closed 2021-07-25 by cpq

vrkansagara, 2021-12-27:
@cpq   #1437 has same issue.