cesanta/mongoose #1320: Segmentation fault in mg_http_creds()

ABH10, 2021-07-21:
- My goal is: To run a HTTP server
- My actions were: No action. The server was listening to random noise/attacks from the Internet.
- My expectation was: No crash
- The result I saw: 
Segmentation fault in mongoose.c, line 1160: 
"if (memcmp(&v->ptr[i], "access_token=", 13) == 0) {" 
caused by too large illegal value of i, again caused by missing range check on v->len in line 1157.
- My question is: Will you fix this?

# Environment
- mongoose version: 7.2
- custom changes?: n
  - If yes, provide the diff below, and the relevant commit:
- build environment:
  - OS: Debian Linux
  - Compiler/IDE: GCC/G++
- relevant example (see https://github.com/cesanta/mongoose/tree/master/examples) is: ?
- relevant documentation topic (see https://cesanta.com/docs/) is: ?

Issue
In the for loop in mongoose.c, line 1159 
"for (i = 0; i < v->len - 13; i++) {"
both i and v->len are size_t variables. If v->len happens to be less than 13, then the loop will continue with i out of bounds and 
"v->ptr[i]" will access illegal memory. 

Proposed change:
In mongoose.c in function mg_http_creds() line 1157:
"  } else if ((v = mg_http_get_header(hm, "Cookie")) != NULL) {"
change to
"  } else if (((v = mg_http_get_header(hm, "Cookie")) != NULL) && (v->len > 13)) {"

cpq, 2021-07-21:
This codepath is already fixed in the HEAD - could you update to the latest version and confirm that the bug is not triggering, please?

ABH10, 2021-07-21:
Thanks, I will update to HEAD.
I have no method for triggering the fault. I was just running a long term test where I saw this fault from analyzing the core dump.

cpq, 2021-07-29:
No response, closing.
Feel free to reopen if HEAD still is an issue.

closed 2021-07-29 by cpq