cesanta/mongoose #1322: Latest mongoose executable detected as Trojan Program:Win32/Wacapew.C!ml

Jaifroid, 2021-07-27:
I'm sure this is a false positive, but you should know that mongoose 7.3.3 is detected by Windows Defender as Program:Win32/Wacapew.C!ml. Screenshot of detection message below. This was detected on a fresh Windows 11 (preview) install. I hope I didn't foolishly "allow" the app...

[image]

Jaifroid, 2021-07-27:
Could it have anything to do with https://github.com/pyinstaller/pyinstaller/issues/5668 (Python)?

cpq, 2021-07-29:
Yes this is a false positive.
We do not create or distribute malware.

closed 2021-07-29 by cpq

Jaifroid, 2021-07-29:
Of course you don't, but maybe there are some steps you can take to ensure it's not flagged like this...