cesanta/mongoose #1323: Mongoose as http client should entertain uris with a colon

ignacionr, 2021-07-27:
- My goal is: To connect as an HTTP client to arbitrary URLs (including those where the host doesn't have a port indication, and the URI contains a colon)
- My actions were: mg_http_connect
- My expectation was: to get a connection to the corresponding host
- The result I saw: ends up failing at DNS (because it parses the url incorrectly)
- My question is: we should cut a host name parsing at either a colon or a slash

# Environment

- mongoose version: trunk
- OS: Ubuntu 21.04
- Compiler/IDE: Clang-12

ignacionr, 2021-07-27:
If you want to wait for next weekend, I can do this. And I promise to add coverage for it.

cpq, 2021-07-29:
@ignacionr Thank you. Could you give an example of the failing URL please?

ignacionr, 2021-07-29:
```https://api.telegram.com/botbotid:bottoken```
If you call connect, it will call get_host, which will call parse_url. I know it's a super easy fix, and I also know that if you encode the colon as %3A it will work, but then again encoding the colon in this situation is not mandatory as per the relevant RFC.

PR #1324 (2021-07-31): allow for in-url colon, when port is not specified
fixes #1323
closed 2021-07-31 by cpq