cesanta/mongoose #1327: Lack of Sec-WebSocket-Protocol in ws_handshake() breaks compatibility with some applications.

volanin, 2021-08-02:
In 7.3, **ws_handshake()** sends the following (it lacks Sec-WebSocket-Protocol):

```
"HTTP/1.1 101 Switching Protocols\r\n"
"Upgrade: websocket\r\n"
"Connection: Upgrade\r\n"
"Sec-WebSocket-Accept: %s\r\n"
```

In 6.18, **mg_ws_handshake()** sent the following (it includes Sec-WebSocket-Protocol) 

```
"HTTP/1.1 101 Switching Protocols\r\n"
"Upgrade: websocket\r\n"
"Connection: Upgrade\r\n");
"Sec-WebSocket-Protocol: %.*s\r\n"
"Sec-WebSocket-Accept: %s%s"
```

Dropping Sec-WebSocket-Protocol in V7 breaks compatibility with some applications, such as jsmpeg (node.js), making the client disconnect immediately.

volanin, 2021-08-02:
Proposed patch attached.

[wsprotocol.patch.gz](https://github.com/cesanta/mongoose/files/6913868/wsprotocol.patch.gz)

cpq, 2021-08-02:
That's WAI.
The API allows to specify extra headers, where those who need protocol set, can do so:

https://cesanta.com/docs/#mg_ws_connect

Closing this.

volanin, 2021-08-02:
Thanks cpq.
I am coming from V6, and didn't realize I could do that. For future reference:

```
case MG_EV_HTTP_MSG:
{
    struct mg_http_message* httpMessage = (struct mg_http_message*)eventData;
    struct mg_str *wsProtocol = mg_http_get_header( httpMessage, "Sec-WebSocket-Protocol" );
    if( wsProtocol )
    {
        mg_ws_upgrade( connection, httpMessage, "Sec-WebSocket-Protocol: %.*s\r\n", (int)wsProtocol->len, wsProtocol->ptr );
    }
    else
    {
        mg_ws_upgrade( connection, httpMessage, nullptr );
    }
}
break;

```

cpq, 2021-08-03:
Actually `ws_upgrade()` function could do that automatically. Could you describe your use case please? That's a server side code, isn't it - and what's the client, out of curiosity?

volanin, 2021-08-03:
I am coding a video streaming websocket server. The client connects to the mongoose server, performs the handshake to upgrade the connection to websockets, and then keeps the connection open to receive video data and display in the browser.

In this specific case, the client is **jsmpeg** (https://github.com/phoboslab/jsmpeg).

When it requests to upgrade the connection to websockets, it sends Sec-WebSocket-Protocol with a list of protocols. When the mongoose server replies to the handshake inside **mg_ws_upgrade()**, it must send this list of supported protocols back to jsmpeg, or else jsmpeg disconnects immediately.

cpq, 2021-08-04:
This commit https://github.com/cesanta/mongoose/commit/c2c6ff11ed11b5c65aa5aa85cfcd78576e352a26
changes `mg_ws_upgrade()` to send `Sec-WebSocket-Protocol` header it it was set by the client.

closed 2021-08-04 by cpq