cesanta/mongoose #1328: Per connection user data was removed in V7.

volanin, 2021-08-02:
Since I am coming from 6.18, this might have some V6 bias.
I am creating a websocket server.

In V6, I can set a "global user data" in **struct mg_mgr** and also a "per connection user data" in **struct mg_connection**. Both of them have _void *user_data_ fields.

In V7, I can still set a "global user data" in **struct mg_mgr** (although setting it directly was removed from the **mg_mgr_init()** API)... but the "per connection user data" was removed from **struct mg_connection**.

It seems to have been replaced by _void *fn_data_, but it's not the same thing, since this field is meant to be set by the function **mg_http_listen()** and as such, I cannot have each connection having its own separate user data pointer.

Of course, in MG_EV_ACCEPT, I can still do **connection->fn_data = malloc( sizeof( struct UserData ))**, but using this field doesn't feel like the right thing to do.

cpq, 2021-08-02:
For accepted connections, it is fine to replace `c->fn_data` - in fact, it's fine to replace it for any connection.

cpq, 2021-08-03:
Closing this - please use `c->fn_data`

closed 2021-08-03 by cpq