cesanta/mongoose #1331: MG_EV_CLOSE event fires before resources are actually freed

GlebPlekhotko, 2021-08-04:
# Suggestion

In the function "**close_conn**" you fire event "**MG_EV_CLOSE**" well before allocated resources are actually freed:
```
static void close_conn(struct mg_connection *c) {
  ...
  mg_call(c, MG_EV_CLOSE, NULL);
  ...
  free(c->recv.buf);
  free(c->send.buf);
  memset(c, 0, sizeof(*c));
  free(c);
}
```

In some cases that might cause issues. At instance in my particular application I use this event to inform the rest of the code that current connection has terminated and resources are ready for reuse. Depending on priorities of the tasks/threads reuse attempt might occur before receive and send buffers are released. In the resources-constrained applications, where heap is quite scarce, that may corrupt execution flow, because no free memory left.

So it seems reasonable to move the point where "**MG_EV_CLOSE**" event fires to the following position:

```
static void close_conn(struct mg_mgr *mgr, struct mg_connection *c) {
  ...
  mg_tls_free(c);
  free(c->recv.buf);
  free(c->send.buf);
  mg_call(c, MG_EV_CLOSE, NULL);
  memset(c, 0, sizeof(*c));
  free(c);
  if (c == mgr->dnsc) mgr->dnsc = NULL;
}
```
In my particular case that works fine.

# Environment

- mongoose version: 7.0
- OS: FreeRTOS 10.3.1
- Compiler/IDE: arm-none-eabi-gcc 9.2.1

cpq, 2021-08-05:
@GlebPlekhotko thank you.
For some use cases, this change won't work.
For example, GET responses without Content-Length header, have no idea what is the real body length, and by spec, read everything until connection is closed. When connection is closed, a buffered message is treated as fully received, and `MG_EV_HTTP_MSG` is fired followed by `MG_EV_CLOSE`. If we deallocate read buffer before that, we loose the message. We can't do that. And usually, send buffer is clear at this stage anyway.
The same goes for HEAD responses.

So this change can't be accepted. What is worth doing though, is adding a relevant comment to `close_conn()`

cpq, 2021-08-05:
@GlebPlekhotko out of curiosity, what is the target board, SDK, and connectivity type?

closed 2021-08-05 by cpq

GlebPlekhotko, 2021-08-05:
Fine, I will keep this change for myself only.

In regard to your question:
- _target board_
Custom hardware based on [nRF52840](https://www.nordicsemi.com/Products/nRF52840) chip + [WINC1500](https://www.microchip.com/en-us/product/ATwinc1500) WiFi module.
- _SDK_
[nRF5 SDK](https://www.nordicsemi.com/Products/Development-software/nrf5-sdk), but I'm actually using only a couple of headers describing hardware resources, like addresses of the peripherals.
-  _connectivity type_
Websocket in both client and server roles, though not simultaneously.

commit 47ce072a (2021-08-05): Fix #1331 - add comment about order of operations in conn_close()