cesanta/mongoose #1332: Mongoose makes it hard to create a server that constantly pushes low latency data to clients.

volanin, 2021-08-05:
This was originally discussed in #903 but I believe it deserves it's own issue ticket. 

It's really hard to create a websocket server in mongoose where the server has to continually push low latency data to the clients without the clients requesting the data. Once a client connects, mongoose will only be able to push data to it when the **select()** inside **mg_mgr_poll()** expires. And since it's impossible to wake up from the **select()** as soon as new data is available, this adds undesirable lag.

My use case:

Right now, I am coding a video streaming websocket server. In the **transcode-thread**, a process (ffmpeg) connects to an RTSP live stream, transcodes the h264 video in realtime to mpeg1 video and writes this data to a buffer. In the **mongoose-thread**, the video data in this buffer must be written to all client connections as fast as possible to keep latency low.

Here is where the problem lies:

- The **mongoose-thread** must read the buffer and call **mg_ws_send()** to send this data to the clients as soon as new data is available, but it can only do this after the **select()** in **mg_mgr_poll()** expires, since there is no way to manually wake up from the **select()**. This will introduce lag up to _timeout milliseconds_.

One good way to solve this problem would be to implement a thread-safe **mg_mgr_wake()** function that could be called from **transcode-thread** immediately after new data is stored in the buffer. This would to wake up the **select()** and the **mongoose-thread** would immediately be free to send the data to the clients. Example code:

```
while( true )
{
    mg_mgr_poll( &mgr, 1000 )
	

    pthread_mutex_lock( &buffer_mutex );
    if( bytes_to_send )
    {
        for( struct mg_connection* c = mgr.conns; c != nullptr; c = c->next )
            if( c->is_accepted )
                mg_ws_send( c, buffer, bytes_to_send, WEBSOCKET_OP_BINARY );

        bytes_to_send = 0;
    }
    pthread_mutex_unlock( &buffer_mutex );
}
```

Transcode-Thread:

```
while( true )
{
    pthread_mutex_lock( &buffer_mutex );
    bytes_to_send = fread( buffer, 1, sizeof( buffer ), ffmpeg_pipe );
    pthread_mutex_unlock( &buffer_mutex );

    mg_mgr_wake( &mgr );  //  This would force mg_mgr_poll() to wake up.
}
```

volanin, 2021-08-05:
One way this could be done is with socketpair(). The function **mg_mgr_wake()** would just write to the first socket, and **mg_mgr_poll()** would just need add the second socket to **select()**, discarding its contents afterwards.

[mg_mgr_wake.patch.gz](https://github.com/cesanta/mongoose/files/6940619/mg_mgr_wake.patch.gz)

This is a proposed patch implementing **mg_mgr_wake()** using **mg_socketpair()** for the 7.3 codebase.
It needs MG_ENABLE_SOCKETPAIR to be true. If not, the **mg_mgr_wake()** simply does nothing.

cpq, 2021-08-05:
This seems to be relevant: https://github.com/cesanta/mongoose/issues/1257

cpq, 2021-08-07:
Please try out at this approach, pushed to a branch `mt`:
- removed `mg_socketpair()`
- added `mg_mkpipe()` and `mg_rmpipe()`.

See https://github.com/cesanta/mongoose/blob/mt/docs/README.md#mg_mkpipe

volanin, 2021-08-10:
It worked perfectly.

But if you don't mind my opinion, I didn't like using this new API.
Here are the reasons that made me think so (remember that I am implementing a server):

1. **mg_mkpipe()** expects a mg_connection as parameter. This feels wrong because when I pass a mg_connection as a parameter, I expect the pipe to work for this mg_connection only.  It gives the false (and error prone) impression that I need to call **mg_mkpipe()** for every new mg_connection that is accepted by my server and then write to each "connection pipe" that I created. But instead, **mg_mkpipe()** affects the mg_mgr. It allows waking up the mg_mgr **select()**, and this affects ALL mg_connections. This function should expect a mg_mgr as parameter.

2. **mg_mkpipe()** has only one purpose: to provide a way to wake up the mg_mgr **select()**... or at least that's what it seems to me; writing to the pipe has no other purpose. This being so, there are too many API details exposed to the developer:
    - There needs to exist only a single pair of pipes per mg_mgr, yet I have to manually declare and store this pair (globally, as storing it in fn_data, as the example does, makes no sense on a server). This pair should live inside the mg_mgr structure, hiding these details from the developer.
     - To write to the pipe, I have to use **mg_http_reply()** and write some made up garbage there. And I also have to know that I must write to pc[0] and not pc[1], or things will not work. All of this just complicates the implementation, and there should be a simple wrapper like **mg_wakeup()** that hides these details from the developer.

3. The newly created pipe is marked with _is_accepted_. This is unexpected, as the pipe is not an accepted connection to my server. It broke my previously working loop, and I had to check mongoose code to find out why:

Original loop:

```
for( struct mg_connection* connection = manager.conns; connection != NULL; connection = connection->next )
	if( connection->is_accepted )
		mg_ws_send( connection, data, size, WEBSOCKET_OP_BINARY );
```

I had to modify it, in order to not cause problems:

```
for( struct mg_connection* connection = manager.conns; connection != NULL; connection = connection->next )
	if( connection->is_accepted && connection->is_websocket )
		mg_ws_send( connection, data, size, WEBSOCKET_OP_BINARY );
```

It all works, yes.
I'd like to suggest keeping this implementation, but simplify the API like so:

**mg_mkpipe( mg_mgr )**  //  Create the pair of pipes if needed, and store them inside struct mg_mgr.
**mg_rmpipe( mg_mgr )**  //  Allow closing the pipes, but it also happens automatically in **mg_mgr_free()**
**mg_wakeup( mg_mgr )**  //  Wrapper to write one byte in the correct pipe and wake up the **mg_mgr select()**

I believe this would make it much clearer and error free.

cpq, 2021-08-11:
The suggested API is designed for a case which is a bit different than yours. In your case, an event manager should be woken up and data broadcasted to many connections. By the way, please use `c->label` as a marker, don't use flags like `c->is_accepted`. Like, when a WS data consumer connection is created, `c->label[0] == 'X';` and then when you broadcast, `for (...) if (c->label[0] == 'X') ...`

Let me rework the API to accomodate for your use case ..

volanin, 2021-08-11:
I didn't know about c->label.
Refactoring the code right now. Thanks.

cpq, 2021-08-11:
Here we go. Refactored:

- https://github.com/cesanta/mongoose/tree/mt/docs#mg_mkpipe-mg_mgr_wakeup - a doc
- https://github.com/cesanta/mongoose/blob/mt/examples/multi-threaded/main.c  an example

cpq, 2021-08-12:
@volanin Wagner I'd love to hear your feedback on a suggested API, please write up when you have a chance.

volanin, 2021-08-14:
Ah, now it's looking really good.

I just refactored my code to use the new API, and everything fits like a charm. I also really enjoyed that I can set a separate event handler for the pipe connection, so when I call **mg_mgr_wakeup()**, I have a specific place to handle pushing data to clients. _(and optionally, I could choose to pass the same event handler to **mg_http_listen()** and **mg_mkpipe()** and have a single event handler for everything...)_

I'd say the only thing I'd personally change is to put _struct mg_connection *pipe_ inside _struct mg_mgr_ to avoid having to  pass it around in _fn_data_ or making it global, and simplify things a little. Especially if I already have custom user data to pass in _fn_data_. In the majority of the cases, implementations will use a single pipe anyway... but having said that, I can see use cases with multiple pipes trigerring different event handlers for different things, so it's really just a personal opinion. The way it's implemented is more powerful, but also a little more complex.

Nothing else to suggest, thanks for implementing this.

cpq, 2021-08-17:
We can't put pipe connection under the event manager's control...

We give pipe connection to another thread. As soon as event manager also accesses it, we need to introduce locking primitives into the API - and that what we'd like to avoid at all costs. Currently, pipe connection is given to another thread, event manager does not have a clue about it, and all locking is done implicitly by the OS when a thread `send()` to a socketpair.

volanin, 2021-08-17:
Ah, I see your point now. I have been convinced.
Again, thanks a lot for implementing this and taking the time to explain everything.

closed 2021-08-17 by volanin