cesanta/mongoose #1338: Udp does not know how to use it?

huanxinmengmeng, 2021-08-17:
I have used UDP to test many methods that cannot communicate properly.  Is there a routine to tell me how to use it?

xiaobai2017, 2021-08-18:
Maybe you can see this function:
test_udp() in master/test/unit_test.c

huanxinmengmeng, 2021-08-18:
I found the reason. It did not use sendto and recvfrom, so the message could not be sent and received normally.

huanxinmengmeng, 2021-08-18:
Finally, I solved it. I made some changes to the source code.
`
static long mg_sock_send(struct mg_connection *c, const void *buf, size_t len) {
  long n = 0; 
  if (c->is_udp) {
    union usa usa;
    usa.sin.sin_addr.s_addr = c->peer.ip;
    usa.sin.sin_port = c->peer.port;
    usa.sin.sin_family = AF_INET;
    n = sendto(FD(c), (char *) buf, len, 0, &usa.sa, sizeof(usa.sin));
  } else {
    n = send(FD(c), (char *) buf, len, MSG_NONBLOCKING);
  }
  return n == 0 ? -1 : n < 0 && mg_sock_would_block() ? 0 : n;
}
`

huanxinmengmeng, 2021-08-18:
`static long mg_sock_send(struct mg_connection *c, const void *buf, size_t len) {
  long n = 0; 
  if (c->is_udp) {
    union usa usa;
    usa.sin.sin_addr.s_addr = c->peer.ip;
    usa.sin.sin_port = c->peer.port;
    usa.sin.sin_family = AF_INET;
    n = sendto(FD(c), (char *) buf, len, 0, &usa.sa, sizeof(usa.sin));
  } else {
    n = send(FD(c), (char *) buf, len, MSG_NONBLOCKING);
  }
  return n == 0 ? -1 : n < 0 && mg_sock_would_block() ? 0 : n;
}`

huanxinmengmeng, 2021-08-18:
Finally, I solved it. I made some changes to the source code.
```
static long mg_sock_send(struct mg_connection *c, const void *buf, size_t len) {
  long n = 0; 
  if (c->is_udp) {
    union usa usa;
    usa.sin.sin_addr.s_addr = c->peer.ip;
    usa.sin.sin_port = c->peer.port;
    usa.sin.sin_family = AF_INET;
    n = sendto(FD(c), (char *) buf, len, 0, &usa.sa, sizeof(usa.sin));
  } else {
    n = send(FD(c), (char *) buf, len, MSG_NONBLOCKING);
  }
  return n == 0 ? -1 : n < 0 && mg_sock_would_block() ? 0 : n;
}
```

closed 2021-08-19 by huanxinmengmeng