cesanta/mongoose #1339: http connection always fialed in mg_tls_handshake function when using openssl

hql420621, 2021-08-20:
hey.thanks for your attention,when i use mongoose.h and mongoos.c file to build a httpserver with openssl in my project on windows，they always crashed when i try to build a connect ,the following is the step i do:
 
- My actions were: 
- 1. run curl comand `curl -v -k -X POST -d '{"type":"log","serverip":"10.43.190.35","agentip":"10.43.190.27","port":443,"heartbeat":60,"product":"las","send":0,"https":1,"replace":1}' https://10.44.239.123:16000/reg
`to get a connection to server,it turns out like this
```
 Initializing NSS with certpath: sql:/etc/pki/nssdb
 skipping SSL peer certificate verification
 NSS error -5961 (PR_CONNECT_RESET_ERROR)
 TCP connection reset by peer
 Closing connection 0
 curl: (35) TCP connection reset by peer
```
2. then i use the wireshark to capture the package,there is a outcome:
```
 .......
116455	9841.864680	10.44.239.123	10.44.237.26	TLSv1.2	1267	Server Hello, Certificate, Server Key Exchange, Server Hello Done
116456	9841.864957	10.44.237.26	10.44.239.123	TCP	60	38594 → 16000 [ACK] Seq=178 Ack=1214 Win=32256 Len=0
116457	9841.866253	10.44.237.26	10.44.239.123	TLSv1.2	147	Client Key Exchange, Change Cipher Spec, Encrypted Handshake Message
116458	9841.868034	10.44.239.123	10.44.237.26	TCP	54	16000 → 38594 [RST, ACK] Seq=1214 Ack=271 Win=0 Len=0
```
it looks like there is always a [rst] send by server,which cause the "tcp connnection reset by peer" problem

3.then i debug my project ,the crash code located at the sinipt of mongoose.cpp
```
void mg_tls_handshake(struct mg_connection* c) {
    struct mg_tls* tls = (struct mg_tls*)c->tls;
    int rc;
    SSL_set_fd(tls->ssl, (int)(long)c->fd);
    rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);
    if (rc == 1) {
        LOG(LL_DEBUG, ("%lu success", c->id));
        c->is_tls_hs = 0;
    }
    else {
        int code;
        ERR_print_errors_fp(stderr);
        code = mg_tls_err(tls, rc);
        if (code != 0) mg_error(c, "tls hs: rc %d, err %d", rc, code);
    }
}
```
the code ‘’SSL_accept(tls-ssl)" always failed ,the it jump to "ERR_print_error_fp(stdeer)",i dont know what cause this every time,very confused by this.may you give me a hand ,thanks very much

# Environment

- mongoose version (e.g. 7.3): _7.3_
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): vs 2019
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8):windows 10

there is a snipt of my code
```
bool HttpsServer::Start()
{
	freopen("log//mylog.log", "a", stdout);
	static const char* s_https_addr = "https://10.43.135.136:16000";  // HTTPS port
	mg_mgr m_mgr;
	mg_log_set("3");
	mg_mgr_init(&m_mgr);
	
	mg_http_listen(&m_mgr, s_https_addr, OnHttpEvent, (void*)1);  // HTTPS listener

	fclose(stdout);
	while (true)
	{
		mg_mgr_poll(&m_mgr, 500);
	}
	mg_mgr_free(&m_mgr);
	return true;
}
void HttpsServer::OnHttpEvent(mg_connection* connection, int event_type, void* event_data, void* fn_data)//
{
	switch (event_type)
	{
	case MG_EV_ERROR:
		freopen("log//mylog.log", "a", stdout);
		printf("%s", event_data);
		writelog((char *)event_data, strlen((char *)event_data));
		//LOG_MSG("heartbeat", LOG_ERROR, "%s", event_data);
		break;
	case MG_EV_HTTP_MSG:
	{
		freopen("log//mylog.log", "a", stdout);
		printf("%s", event_data);

		writelog((char*)event_data, strlen((char*)event_data));
		struct mg_http_message* http_req = (struct mg_http_message*)event_data;
		HandleEvent(connection, http_req);
		fclose(stdout);
		break;
	}
	case MG_EV_ACCEPT:
		
		//writelog("connect...\n", 100);
		if (fn_data != NULL) {

				struct mg_tls_opts opts;
				memset(&opts, 0, sizeof(opts));
				opts.cert = "server.pem";
				opts.certkey = "server.key";

				freopen("log//mylog.log", "a", stdout);
				printf("init ssl\n");

				mg_tls_init(connection, &opts);
				fclose(stdout);

		}
		
		break;
	default:
		break;
	}
}

hql420621, 2021-08-20:
i use the curl in server,it turns out 
```
C:\Users\hequanlin> curl -v -k -X POST -d '{"type":"log","serverip":"10.43.190.35","agentip":"10.43.190.27","port":443,"heartbeat":60,"product":"las","send":0,"https":1,"replace":1}' https://10.43.135.136:16000/reg
Note: Unnecessary use of -X or --request, POST is already inferred.
*   Trying 10.43.135.136...
* TCP_NODELAY set
* Connected to 10.43.135.136 (10.43.135.136) port 16000 (#0)
* schannel: SSL/TLS connection with 10.43.135.136 port 16000 (step 1/3)
* schannel: disabled server certificate revocation checks
* schannel: verifyhost setting prevents Schannel from comparing the supplied target name with the subject names in server certificates.
* schannel: using IP address, SNI is not supported by OS.
* schannel: sending initial handshake data: sending 147 bytes...
* schannel: sent initial handshake data: sent 147 bytes
* schannel: SSL/TLS connection with 10.43.135.136 port 16000 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with 10.43.135.136 port 16000 (step 2/3)
* schannel: failed to receive handshake, SSL/TLS connection failed
* Closing connection 0
* schannel: shutting down SSL/TLS connection with 10.43.135.136 port 16000
* Send failure: Connection was reset
* schannel: failed to send close msg: Failed sending data to the peer (bytes written: -1)
* schannel: clear security context handle
curl: (35) schannel: failed to receive handshake, SSL/TLS connection failed
```

cpq, 2021-08-20:
Could you provide a crash backtrace please?

hql420621, 2021-08-23:
thanks for replying. i didn't get crash backtrace rather than modify `mongoose.c`by adding extra code under  `rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl)`after reading offIcal doc about [ssl_accept](https://www.openssl.org/docs/man1.1.1/man3/SSL_accept.html). like the following code,i'm not sure if there is neccssary to repetly calling `SSL_accept(tls->ssl)` in no-blocking mongoose,but it indeed works .
```
void mg_tls_handshake(struct mg_connection* c) {
    struct mg_tls* tls = (struct mg_tls*)c->tls;
    int rc;
    SSL_set_fd(tls->ssl, (int)(long)c->fd);
    rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);
    // add extra snippet
    int ret_ = SSL_get_error(tls->ssl, rc);
    switch (ret_) {
    case SSL_ERROR_WANT_READ:
        rc=c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);
    break;
    case SSL_ERROR_WANT_WRITE:
        break;
    defalut:
        break;
    }
    //
    if (rc == 1) {
        LOG(LL_DEBUG, ("%lu success", c->id));
        c->is_tls_hs = 0;
    }
    else {
        int code;
        ERR_print_errors_fp(stderr);
        code = mg_tls_err(tls, rc);
        if (code != 0) mg_error(c, "tls hs: rc %d, err %d", rc, code);
    }
}
```

cpq, 2021-08-23:
@hql420621 Thank you.

That is hard to accept, because the logic behind the change is unclear.
Would you consider getting a crash backtrace to get to the root of the issue and make a proper fix?

hql420621, 2021-08-24:
okay,i will give it shot to dig out what cause this.

cpq, 2021-08-29:
Pushed https://github.com/cesanta/mongoose/commit/e3589577ed3a130b82d0e81252a9c34a76427c7c
Could you please re-test using the latest version?

hql420621, 2021-08-30:
thank you very much,i will try it with lastest version

hql420621, 2021-08-30:
it works with lastest version,but i dont figure out there is prominent difference between before version. and the wird thing is it suddenly works as well when i go back to the before version（7.3）.

cpq, 2021-08-30:
The difference is the order of calls, `SSL_get_error()` gets called first now.

cpq, 2021-09-04:
Ok, closing this for now - feel free to reopen if required.

closed 2021-09-04 by cpq