cesanta/mongoose #1340: HTTP-Client: Chunked response bug?

lnksz, 2021-08-20:
- My goal is: Get a response from an HTTP server, where I  cannot know whether the respond will be chunked or not.
- My actions were: test with the example/http-{client,streaming-client}
- My expectation was: based on the [unit tests](https://github.com/cesanta/mongoose/blob/master/test/unit_test.c#L1267), I expect that if I do not delete the chunks upon `MG_EV_HTTP_CHUNK`, then upon `MG_EV_HTTP_MSG` I should see the concatenated chunks.
- The result I saw: With both examples, upon `MG_EV_HTTP_MSG` the body was empty.
- My question is: 

# Environment

- mongoose version (e.g. 7.3): 7.3 (checked out from master as `git checkout 7.3`)
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): GCC 11.1.0
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): Linux 5.4
- Target hardware/board (e.g. STM32F411): x86_64
- Connectivity chip/module (e.g. WizNet W5500): localhost
- Target SDK (e.g. STM32CubeF1 1.8.4): -

I started to analyze it because in the "wild" I got a response from an HTTP server, for which my until now working (non-chunked) client logic started to fail because of empty body:

```
HTTP/1.1 200 OK
Transfer-Encoding: chunked
Content-Type: application/json; charset=utf-8
Server: Microsoft-HTTPAPI/2.0
Date: Fri, 20 Aug 2021 12:58:35 GMT
 

15
"2021-08-20 14:58:35"
0
```

To reproduce it I stub this with a minimal go server (can be built and run by `go run main.go`):
```go
package main

import (
	"fmt"
	"net/http"
	"net/http/httputil"
	"log"
)

func main() {
	http.HandleFunc("/", handlerChunk)
	log.Fatal(http.ListenAndServe(":8080", nil))
}

func handlerChunk(w http.ResponseWriter, r *http.Request) {
	flusher, ok := w.(http.Flusher)
	if !ok {
	    panic("expected http.ResponseWriter to be an http.Flusher")
	}

	fmt.Fprintf(w, "\"2021-08-19 18:31:00\"")
	flusher.Flush() // Trigger "chunked" encoding and send a chunk...
	fmt.Fprintf(w, "")
	flusher.Flush()
}
```

The examples (with minimal changes) output:

### http-client

```diff
diff --git a/examples/http-client/main.c b/examples/http-client/main.c
index 9400bf2c..b87c1d8f 100644
--- a/examples/http-client/main.c
+++ b/examples/http-client/main.c
@@ -12,7 +12,7 @@
 #include "mongoose.h"
 
 // The very first web page in history. You can replace it from command line
-static const char *s_url = "http://info.cern.ch//foo";
+static const char *s_url = "http://127.0.0.1:8080";
 
 // Print HTTP response and signal that we're done
 static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
@@ -28,14 +28,14 @@ static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
 
     // Send request
     mg_printf(c,
-              "GET %s HTTP/1.0\r\n"
+              "GET %s HTTP/1.1\r\n"
               "Host: %.*s\r\n"
               "\r\n",
               mg_url_uri(s_url), (int) host.len, host.ptr);
   } else if (ev == MG_EV_HTTP_MSG) {
     // Response is received. Print it
     struct mg_http_message *hm = (struct mg_http_message *) ev_data;
-    printf("%.*s", (int) hm->message.len, hm->message.ptr);
+    printf("Response:\n|%.*s|\n", (int) hm->message.len, hm->message.ptr);
     c->is_closing = 1;         // Tell mongoose to close this connection
     *(bool *) fn_data = true;  // Tell event loop to stop
   } else if (ev == MG_EV_ERROR) {
```

Output:
```
./example 
2021-08-20 19:10:12    sock.c:374:mg_connect     1 -> http://127.0.0.1:8080
Response:
|HTTP/1.1 200 OK
Date: Fri, 20 Aug 2021 19:10:12 GMT
Content-Type: text/plain; charset=utf-8
Transfer-Encoding: chunked

|
2021-08-20 19:10:12    sock.c:290:close_conn     1 closed
2021-08-20 19:10:12  I net.c:122:mg_mgr_free     All connections closed
```
Wireshark, response:
```
Hypertext Transfer Protocol
    HTTP/1.1 200 OK\r\n
    Date: Fri, 20 Aug 2021 19:15:09 GMT\r\n
    Content-Type: text/plain; charset=utf-8\r\n
    Transfer-Encoding: chunked\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 0.001000138 seconds]
    [Request in frame: 4]
    [Request URI: http://127.0.0.1/]
    HTTP chunked response
        Data chunk (21 octets)
            Chunk size: 21 octets
            Data (21 bytes)
            Chunk boundary: 0d0a
        End of chunked encoding
            Chunk size: 0 octets
        \r\n
    File Data: 21 bytes
0000   48 54 54 50 2f 31 2e 31 20 32 30 30 20 4f 4b 0d   HTTP/1.1 200 OK.
0010   0a 44 61 74 65 3a 20 46 72 69 2c 20 32 30 20 41   .Date: Fri, 20 A
0020   75 67 20 32 30 32 31 20 31 39 3a 31 35 3a 30 39   ug 2021 19:15:09
0030   20 47 4d 54 0d 0a 43 6f 6e 74 65 6e 74 2d 54 79    GMT..Content-Ty
0040   70 65 3a 20 74 65 78 74 2f 70 6c 61 69 6e 3b 20   pe: text/plain; 
0050   63 68 61 72 73 65 74 3d 75 74 66 2d 38 0d 0a 54   charset=utf-8..T
0060   72 61 6e 73 66 65 72 2d 45 6e 63 6f 64 69 6e 67   ransfer-Encoding
0070   3a 20 63 68 75 6e 6b 65 64 0d 0a 0d 0a            : chunked....
```

### http-streaming-client

```diff
diff --git a/examples/http-streaming-client/main.c b/examples/http-streaming-client/main.c
index 8e2d5de9..24dc40d7 100644
--- a/examples/http-streaming-client/main.c
+++ b/examples/http-streaming-client/main.c
@@ -16,7 +16,7 @@
 #include "mongoose.h"
 
 // The very first web page in history. You can replace it from command line
-static const char *s_url = "http://info.cern.ch";
+static const char *s_url = "http://127.0.0.1:8080";
 
 // Print HTTP response and signal that we're done
 static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
@@ -38,13 +38,13 @@ static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
               mg_url_uri(s_url), (int) host.len, host.ptr);
   } else if (ev == MG_EV_HTTP_CHUNK) {
     struct mg_http_message *hm = (struct mg_http_message *) ev_data;
-    LOG(LL_INFO, ("%.*s", (int) hm->chunk.len, hm->chunk.ptr));
+    LOG(LL_INFO, ("CHUNK:\n|%.*s|", (int) hm->chunk.len, hm->chunk.ptr));
     mg_http_delete_chunk(c, hm);
     if (hm->chunk.len == 0) *(bool *) fn_data = true;  // Last chunk
   } else if (ev == MG_EV_HTTP_MSG) {
     // Response is received. Print it
     struct mg_http_message *hm = (struct mg_http_message *) ev_data;
-    LOG(LL_INFO, ("%.*s", (int) hm->body.len, hm->body.ptr));
+    LOG(LL_INFO, ("BODY:\n|%.*s|", (int) hm->body.len, hm->body.ptr));
     c->is_closing = 1;         // Tell mongoose to close this connection
     *(bool *) fn_data = true;  // Tell event loop to stop
   } else if (ev == MG_EV_ERROR) {
```
```
./example 
2021-08-20 19:09:59    sock.c:374:mg_connect     1 -> http://127.0.0.1:8080
2021-08-20 19:09:59  I main.c:41:fn              CHUNK:
|"2021-08-19 18:31:00"|
2021-08-20 19:09:59    sock.c:229:read_conn      1   01000000000010      97 -1/927 err 115
2021-08-20 19:09:59  I main.c:47:fn              BODY:
||
2021-08-20 19:09:59    sock.c:290:close_conn     1 closed
2021-08-20 19:09:59  I net.c:122:mg_mgr_free     All connections closed
```

I can see that with e84ea91a4acde36e471d718cb5f5a18935dbcdae and 1b5cc341a5da4c0a70d4205702e10dd3dcf6140a the streaming feature is relatively new in v7 and this is possibly a bug? I tried the unit test, and it was passing, but as there is mongoose on both sides with the stub-server something goes unnoticed? Or are the posted behaviors as expected?

lnksz, 2021-08-21:
I cannot currently retest, but this line looks like a copy and paste bug:
        size_t n = cl < ll + 2 ? 0 : cl2 - ll - 2;
I guess what was wanted is:
        size_t n = cl2 < ll + 2 ? 0 : cl2 - ll - 2;

cpq, 2021-08-21:
@lnksz that definitely looks like a copy/paste bug. Looking how to adjust a unit test to catch it..

lnksz, 2021-08-22:
I would try to have chunks in the test, which have a payload length taking up more then a single hex digit. Like that `ll` would be at least 4. But with the current `ll=3` and the terminator chunk length 5, the test sample accidentally took the expected branch.

I can prepare a PR on Monday.

cpq, 2021-08-22:
@lnksz Appreciate your effort

lnksz, 2021-08-23:
@cpq I added the bugfix PR.

I've split the fix in 2 commits (like in TDD steps), but I can squash them if desired.

I've signed CLA before.

I was thinking about a rewrite of `walkchunks` and `get_chunk_length`, would that be something you would consider merging? (considering chunk extensions + some refactoring)

During debugging, I kind of struggled with the indentation, short variable names. (Although I am not a fan of polluting the code with a lot of comments, I would add a link to the chunk formats.) I think that could be more helpful in the future, then the current commented-out `LOG` macros, from which some have references to non-defined vars.

I understand that style can be highly subjective, so do you have a format/standard guide you want PRs to be conformant to? (I have seen the minimalist.clang-format file, but for e.g. declaring variables and variable naming that does not define anything. Do you e.g. have a C standard version you want to be compatible with?)

closed 2021-08-23 by cpq

cpq, 2021-08-25:
@lnksz thank you!

1. Refactoring `walkchunks()` would make sense to avoid code duplication
2. The style guide is per `clang-format`. The config is in the repo root, `.clang-format`, which is Google style with those few tweaks. Installing a clang-format utility and appropriate editor plugin should do the job.

PR #1341 (2021-08-23): Fix #1340: Long HTTP Chunks
Fix a copy/paste error, which resulted in only short (<0x10) chunks being handled as expected.

Fixes #1340 .

Out of curiosity I cut out the relevant parts of the code to godbolt.org and enabled PVS Studio to see whether their (excellent) copy-paste detector would throw a warning: https://godbolt.org/z/bP1P93faE

But it didn't :( 
Anyway, it would be interesting to register this project to their Open Source free service, in other projects I have seen wonders from this tool:
https://pvs-studio.com/en/blog/posts/0614/
commit 9b37ac38 (2021-08-23): Merge pull request #1341 from lnksz/bugfix-1340-http_chunked

Fix #1340: Long HTTP Chunks