cesanta/mongoose #1347: examples/http-server/main.c文件中有小问题

wucq-jj, 2021-09-06:
- My goal is: _执行./example  2312312   得到usage信息_
- My actions were: _./example 23112_
- My expectation was: _正确的usage_
- The result I saw: _provide logs or description_
- My question is: _static void usage(const char *prog) {
 31   fprintf(stderr,
 32           "Mongoose v.%s\n"
 33           "Usage: %s OPTIONS\n"
 34           "  -H yes|no - enable traffic hexdump, default: '%s'\n"
 35           "  -S GLOB   - glob pattern for SSI files, default: '%s'\n"
 36           "  -d DIR    - directory to serve, default: '%s'\n"
 37           "  -l ADDR   - listening address, default: '%s'\n"
 38           "  -v LEVEL  - debug level, from 0 to 4, default: '%s'\n",
 39           MG_VERSION, prog, s_debug_level, s_enable_hexdump, s_ssi_pattern,
 40           s_root_dir, s_listening_address);
 41   exit(EXIT_FAILURE);
 42 }    如果执行examples 文件时 输入一个错误的OPTIONS  fprintf()打印信息错误 错位了 需要修改_

# Environment

- mongoose version (e.g. 7.3): _7.3_
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): __
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): _?_
- Target hardware/board (e.g. STM32F411): _?_
- Connectivity chip/module (e.g. WizNet W5500): _?_
- Target SDK (e.g. STM32CubeF1 1.8.4): _?_

wucq-jj, 2021-09-06:
hello
My goal   ./mongoose7.3/examples/http-server/example 3213123
I want to execution the usage function to get ,  i just found it will return an incorrect result.

cpq, 2021-09-06:
Please learn how to report an issue.

closed 2021-09-06 by cpq