cesanta/mongoose #1349: Using `mg_serve_http_dir` for non-root URIs

Bklyn, 2021-09-15:
- My goal is: to serve files from a directory over HTTP from a non-root path
- My actions were: use mg_http_serve_dir with opts.path = "path/to/files" but a URI of "/elsewhere"
- My expectation was: directory / files served
- The result I saw: 404 error
- My question is: Can I serve files from a URI that is not the root of my server?

# Environment

- mongoose version (e.g. 7.3): 7.3-80-gdf848973
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): gcc 10.2
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): Linux
- Target hardware/board (e.g. STM32F411): x86_64
- Connectivity chip/module (e.g. WizNet W5500):
- Target SDK (e.g. STM32CubeF1 1.8.4):

Imagine I keep the files I want to serve in a directory named `data` but I want to serve these from a URI on my server that starts with `/static/`.  If I do this, by calling `mg_http_serve_dir` only when `mg_http_match_uri(hm, "/static/#")` is true and passing `opts.root_dir="data"`, I get a 404 / "Not found" response.  I believe `uri_to_path` is the culprit here ("Temporarily append URI to the root dir..."), since there is no `data/static/...` in my case.

I am able to get this working more or less by rewriting the URI in the message that is passed to `mg_http_serve_dir` and stripping off the `/static/` prefix.  Is there another option?

cpq, 2021-09-17:
@Bklyn if `opts.root_dir` supports an optional comma-separated list of uri/dir pairs, would that suffice? Imagine this:

```c
opts.root_dir = "web_root,/static/=/some/other/path";
```

Bklyn, 2021-09-17:
So this would serve URIs that start with `/static/` from the local path `/some/other/path`, and everything else from the directory `web_root`?  That seems fine!

cpq, 2021-09-20:
Pushed https://github.com/cesanta/mongoose/commit/2139fbc4b78da492a468baedaf960964f88d7131

Please test and feedback

cpq, 2021-09-23:
Closing this, feel free to reopen if required.

closed 2021-09-23 by cpq