cesanta/mongoose #1352: mqtt-server'sample can't subscribe topics

tercelcn, 2021-09-17:
- My goal is: mqtt-server work!
- My actions were: mongoose-master/examples/mqtt-server$ make
- The result I saw: 
- ./example 
2021-09-17 06:30:44 2 sock.c:473:mg_listen       1 accepting on mqtt://0.0.0.0:1883 (port 1883)
2021-09-17 06:31:00 2 main.c:48:fn               SUB 0x4 [testtopic/#]
2021-09-17 06:31:06 2 main.c:48:fn               SUB 0x4 [testtopic/#]

- My question is: client can't subscribe topic

# Environment

- mongoose version (e.g. 7.3): 7.3
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): 5.4.0-6ubuntu1~16.04.12
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): ubuntu 16.04
- Target hardware/board (e.g. STM32F411): PC
- Connectivity chip/module (e.g. WizNet W5500): pc
- Target SDK (e.g. STM32CubeF1 1.8.4): pc

cpq, 2021-09-17:
Mongoose MQTT implements MQTT version 3.x. Are you using MQTT client version 5.x ?

cpq, 2021-09-17:
Added version check in https://github.com/cesanta/mongoose/commit/9b6b1a68f49a2807175af07df58a8c277f2f4204

tercelcn, 2021-09-18:
> Mongoose MQTT implements MQTT version 3.x. Are you using MQTT client version 5.x ?

I use MQTTX, mqtt version is 3.1.1.

cpq, 2021-09-18:
Added suback in https://github.com/cesanta/mongoose/commit/2b48e8b03a918987a4d3d671d439d402dff033bd, please feedback

cpq, 2021-09-18:
Tested with hivemq client, 

```sh
$ mqtt sub -h 127.0.0.1 -t foo -i bla -V 3 -q 2
```

```sh
$ mqtt pub -h 127.0.0.1 -t foo -m hello -V 3
```

tercelcn, 2021-09-19:
> Tested with hivemq client,
> 
> ```shell
> $ mqtt sub -h 127.0.0.1 -t foo -i bla -V 3 -q 2
> ```
> 
> ```shell
> $ mqtt pub -h 127.0.0.1 -t foo -m hello -V 3
> ```

Great! Client can sub topic with QOS 0, but can't unsubscribe.

cpq, 2021-09-19:
It was never meant to - the unsubscribe path was not in the example.
Feel free to add it - it'd be quite similar to the subscribe codepath.
Please note that this example does not meant to be a fully compliant MQTT server.

Feel free close the issue, considering the original problem is solved.

cpq, 2021-09-21:
Closing this, and thanks for the report.

closed 2021-09-21 by cpq