cesanta/mongoose #1353: IPv4-mapped address

gvanem, 2021-09-18:
- My goal is:  Connect using a IPv4-mapped address like:
  `mg_connect (&mgr, "tcp://[::ffff:10.0.0.21]:2121", handler, arg);`
- My expectation was: Connection should be tried without any DNS resolution.
- The result I saw: Mongoose tries a DNS lookup which off-course fails.
- My question is: Is IPv4-mapped addresses not supported?

I've added `-DMG_ENABLE_IPV6=1` to my Makefile. So IPv6 should be supported. But it's not working with 
the above address form. Instead this is the network trace of the DNS-query:
```
  * 4.179086 sec: externals/mongoose.c(2482) (mg_mgr_init+46):
    WSAStartup (2.2) --> No error.

  * 4.192850 sec: externals/mongoose.c(3152) (mg_connect_resolved+164):
    socket (AF_INET, SOCK_DGRAM, IPPROTO_IP) --> 908.

  * 4.197397 sec: externals/mongoose.c(3161) (mg_connect_resolved+534):
    connect (908, 8.8.8.8:53, fam AF_INET) --> No error.

  * 4.205761 sec: externals/mongoose.c(2940) (mg_send+155):
    send (908, 0x00CFFC88, 34, 0) --> 34 bytes.
    0000: 00 01 01 00 00 01 00 00 00 00 00 00 09 3A 3A 66  .............::f
    0010: 66 66 66 3A 31 30 01 30 01 30 02 32 31 00 00 01  fff:10.0.0.21...
    0020: 00 01                                            ..

  * 4.236212 sec: externals/mongoose.c(3401) (mg_mgr_poll+3678):
    recvfrom (908, 0x0B82C020, 2048, 0, 8.8.8.8:53) --> 109 bytes.
    0000: 00 01 81 83 00 01 00 00 00 01 00 00 09 3A 3A 66  ..üâ.........::f
    0010: 66 66 66 3A 31 30 01 30 01 30 02 32 31 00 00 01  fff:10.0.0.21...
    0020: 00 01 00 00 06 00 01 00 01 51 7F 00 40 01 61 0C  .........Q..@.a.
    0030: 72 6F 6F 74 2D 73 65 72 76 65 72 73 03 6E 65 74  root-servers.net
    0040: 00 05 6E 73 74 6C 64 0C 76 65 72 69 73 69 67 6E  ..nstld.verisign
    0050: 2D 67 72 73 03 63 6F 6D 00 78 77 69 D8 00 00 07  -grs.com.xwi╪...
    0060: 08 00 00 03 84 00 09 3A 80 00 01 51 80           ....ä..:Ç..QÇ
```

And the `MG_EV_ERROR`  event: `DNS lookup failed`.

# Environment

- mongoose version: 7.4
- Compiler/IDE: MSVC-2019, clang-cl
- Target OS: Win-10

cpq, 2021-09-18:
@gvanem ipv4-mapped form is not yet supported by the `mg_aton6()` function that parses ipv6 address. Should not be hard to recognize though .. Stay tuned, a change is coming.

cpq, 2021-09-18:
Pushed https://github.com/cesanta/mongoose/commit/a7afea0963c308a1cf0c28f522d4774137b45437, please try HEAD.

gvanem, 2021-09-19:
It works fine. But when I use `mg_url_host()` on such a  string, it's a bit annoying to have to add the brackets (`[%s]`) again in `mg_connect()`.

cpq, 2021-09-19:
Could you share the code snippet you're using please

cpq, 2021-09-21:
It makes sense to follow what others do, and include brackets into the parsed host.

Pushed https://github.com/cesanta/mongoose/commit/90a131bd294f3d3c04dffbac12289d95d6be1743

Closing this, and thank you.

closed 2021-09-21 by cpq