cesanta/mongoose #1357: simple http server compiler errors

Slluxx, 2021-09-23:
- My goal is: to run a simple http server that serves static sites
- My actions were: using/trying the example code as in docs and in the examples folder
- My expectation was: it works
- The result I saw: `argument of type "void *" is incompatible with parameter of type "mg_http_message *"`
- My question is: whats the correct way of serving files?

```c++
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  struct mg_http_serve_opts opts = {.root_dir = "romfs:/www"};   // Serve local dir
  if (ev == MG_EV_HTTP_MSG) mg_http_serve_dir(c, ev_data, &opts);
}
```

`mg_http_serve_dir` seems to expect `mg_http_message *hm` as second argument but my compiler says thats not the case.

# Environment

- mongoose version (e.g. 7.3): 7.3
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): aarch64-none-elf-g++
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): Horizon OS
- Target hardware/board (e.g. STM32F411): Nintendo Switch
- Connectivity chip/module (e.g. WizNet W5500): _?_
- Target SDK (e.g. STM32CubeF1 1.8.4): _?_

nliviu, 2021-09-23:
You have to cast `ev_data` to `mg_http_message *`.

Slluxx, 2021-09-23:
yeah i just did that a second ago and now i have a different problem.

the compiler wants to use `<mongoose_custom.h>` because `"MG_ARCH is not specified and we couldn't guess it."`
I am cross-compiling a homebrew to a gameconsole so im not sure how to proceed from here.

cpq, 2021-09-24:
Please read https://github.com/cesanta/mongoose/tree/master/docs#custom-build

closed 2021-09-24 by cpq