cesanta/mongoose #1361: unit_test.exe on Windows

gvanem, 2021-09-24:
I modified `test/unit_test.c` a bit to continue on a failure:
```c
static int s_num_tests = 0;
static int s_num_failed = 0;

#define ASSERT(expr)                                            \
  do {                                                          \
    s_num_tests++;                                              \
    if (!(expr)) {                                              \
      printf("FAILURE %s:%d: %s\n", __FILE__, __LINE__, #expr); \
      s_num_failed++;                                           \
   /* exit(EXIT_FAILURE); */                                    \
    }                                                           \
  } while (0)
```

This gave me 15 failures (out of 566 total). Granted, some test could depend on the result of a previous failed test.
But most failure seems to me to depend on line-endings. E.g. on line 988:
```
ASSERT(mg_strcmp(hm.body, mg_str("is disease.\n")) == 0);
```
fails since the test assumes `test/data/range.txt` has Unix line-ending. Doing a `dos2unix test/data/range.txt` first, 
this test succeeded. I've not bothered to examine all the other failures.

Just a FYI.

gvanem, 2021-09-24:
*Update:* With the added `-DMG_ENABLE_PACKED_FS=1` and the generated `test/packed_fs.c`, I'm down to 5 failures:
```
FAILURE test/unit_test.c:917: (v = mg_http_get_header(&req, "h")) == NULL
FAILURE test/unit_test.c:539: cmpbody(buf, "Ñ”\n") == 0
FAILURE test/unit_test.c:581: cmpbody(buf, "this is index\n" "this is nested\n\n" "this is f1\n\n\n\n" "recurse\n\n" "recurse\n\n" "recurse\n\n" "recurse\n\n" "recurse\n\n") == 0
FAILURE test/unit_test.c:616: errored == 7
FAILURE test/unit_test.c:763: ok == 200
```

Something else than line-endings?

cpq, 2021-09-24:
Windows .exe are verufied by Github actions on every commit and they do not fail.

closed 2021-09-24 by cpq

gvanem, 2021-09-24:
> Windows .exe are verufied by Github actions on every commit and they do not fail.

Since that seems to run with Docker and Wine, they probably don't. No surprise since that implies Unix line-endings.