cesanta/mongoose #1367: redirection

dicosky, 2021-10-03:
Hi
I have a question about redirection : 
if the webserver accept 2 ports : (27015 http) and (27016 https).
the client can use it like this : http://www.mysite.com:27015 for HTTP  and https://www.mysite.com:27016 for HTTPS. 
but what if the client call the site like this http://www.mysite.com:27016 ? the client use http with port that is created just for https. 
how i can redirect the client to correct port ?

```
static void ev_handler(struct mg_connection *connection, int ev, void *ev_data, void *fn_data)
{
    if(ev == MG_EV_CONNECT)
    {

    }
    else if(ev == MG_EV_ACCEPT)
    {

        if(fn_data != NULL)
        {
            struct mg_tls_opts opts =
            {
                .cert = ssl_cert.c_str(),
                .certkey = ssl_key.c_str(),
            };
            mg_tls_init(connection, &opts);
        }

    }
    else if(ev == MG_EV_WS_MSG)
    {

    }
    else if(ev == MG_EV_HTTP_MSG)
    {

    }
    else if(ev == MG_EV_CLOSE)
    {

    }
    (void) fn_data;
}
```

thanks

cpq, 2021-10-04:
The short answer - you cannot. As soon as you wrap a connection into TLS, a non-TLS data would raise an exception and a connection will close.

dicosky, 2021-10-05:
> The short answer - you cannot. As soon as you wrap a connection into TLS, a non-TLS data would raise an exception and a connection will close.

Any idea to check the connection before wrap it into TLS ?

dicosky, 2021-10-05:
i fixed it , thanks

closed 2021-10-05 by cpq