cesanta/mongoose #1369: Why doesn't connection close automatically if DNS request fails/times out?

GlebPlekhotko, 2021-10-05:
# Environment
- mongoose version (e.g. 7.3): 7.0
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): arm-none-eabi-gcc 9.2.1
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): FreeRTOS 10.3.1
- Target hardware/board (e.g. STM32F411): nRF52840
- Connectivity chip/module (e.g. WizNet W5500): ATWINC1500
- Target SDK (e.g. STM32CubeF1 1.8.4): None

# Desription
When creating a connection (websocket in my case), mongoose automatically creates ancillary one to perform DNS request if a given URL does not contain explicit address of a remote. If DNS completes fine primary connection starts operating duly.

If DNS request fails (e.g. timeout due to lost Internet connection), _mongoose_ detects that, but wipes out **only** DNS connection itself, leaving the one which caused its creating pending for the resolve which will never happen.

I believe it's reasonable to close all connections pending for resolve in this case or at least inform _mongoose_ user (software I mean) about this kind of failure.

In my particular case I fixed this as following and shallow testing says it's fine:
```
static void dns_cb(struct mg_connection *c, int ev, void *ev_data,
                   void *fn_data) {
  struct dns_data *d, *tmp, **head = (struct dns_data **) &c->pfn_data;
  if (ev == MG_EV_POLL) {
     ...
  } else if (ev == MG_EV_READ) {
     ...
  } else if (ev == MG_EV_CLOSE) {
    for (d = *head; d != NULL; d = tmp) {
      >>> if (d->c->is_resolving) d->c->is_closing = 1; <<<
      tmp = d->next;
      mg_dns_free(head, d);
    }
  }
}
```

Is it ok? Looking forward for your comment.

Best regards.

cpq, 2021-10-05:
@GlebPlekhotko you're working with 7.0. How about the latest codebase? Does it have the same issue?

GlebPlekhotko, 2021-10-05:
I've pulled the latest version of your library and peeked **dns_cb** function. And there are the same three lines of code:
'''
static void dns_cb(struct mg_connection *c, int ev, void *ev_data,
                   void *fn_data) {
  struct dns_data *d, *tmp;
  if (ev == MG_EV_POLL) {
    ...
  } else if (ev == MG_EV_READ) {
    ...
  } else if (ev == MG_EV_CLOSE) {
    for (d = s_reqs; d != NULL; d = tmp) {
      tmp = d->next;
      mg_dns_free(d);
    }
  }
  (void) fn_data;
}
'''
So it seems, behavior is the same.