cesanta/mongoose #1392: Add build options to disable mqtt and sntp

jcorporation, 2021-10-31:
- My goal is: build a binary without mqtt and sntp 
- My actions were: removed it manually from makefile 
- My expectation was: an existing build option
- The result I saw: mongoose works flawlessly without mqtt and sntp source/header files
- My question is: can you please  add this two build options?

cpq, 2021-10-31:
Did you consider `-ffunction-sections -fdata-sections -Wl,--gc-sections` build option? This makes a compiler to insert each function into its own ELF section. Then, a linker performs garbage collection (GC) of the unused sections. Thus, if you don't call MQTT routines, they won't be in your binary.

If you aim for a minimal binary, you should do it anyways, to remove unused functions from other places, too. There is no need for an extra build flag. Please verify that, and feedback.

jcorporation, 2021-10-31:
I don’t considered your proposal but I will test it.

I want to keep my patchset for mongoose as small as possible.

jcorporation, 2021-11-01:
It works like you described it. Thanks for the fast response.

closed 2021-11-01 by jcorporation