cesanta/mongoose #1401: docker: Error response from daemon: the working directory 'mdashnet/cc2' is invalid, it needs to be an absolute path.

mailinglists35, 2021-11-07:
````
# export IPV6=0
# make linux
cc -O3 -g3 -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -DMG_MAX_HTTP_HEADERS=7 -DMG_ENABLE_LINES -DMG_ENABLE_PACKED_FS=1   -DMG_ENABLE_MBEDTLS=1 -I/usr/include -I/usr/include -fsanitize=address,undefined test/pack.c -o pack
./pack Makefile src/fs.h src/ssi.h test/fuzz.c test/data/a.txt > test/packed_fs.c
docker run --rm -e Tmp=. -e WINEDEBUG=-all -v : -w  mdashnet/cc2 gcc mongoose.c test/unit_test.c test/packed_fs.c -O3 -g3 -W -Wall -Werror -Wshadow -Wdouble-promotion -fno-common -Wconversion -Isrc -I. -DMG_MAX_HTTP_HEADERS=7 -DMG_ENABLE_LINES -DMG_ENABLE_PACKED_FS=1   -DMG_ENABLE_MBEDTLS=1 -I/usr/include -I/usr/include -fsanitize=address,undefined -L/usr/lib -lmbedtls -lmbedcrypto -lmbedx509 -o unit_test_gcc
docker: Error response from daemon: the working directory 'mdashnet/cc2' is invalid, it needs to be an absolute path.
See 'docker run --help'.
Makefile:104: recipe for target 'linux' failed
make: *** [linux] Error 125
````

````
# docker version
Client:
 Version:           20.10.7
 API version:       1.41
 Go version:        go1.13.8
 Git commit:        20.10.7-0ubuntu5~18.04.2
 Built:             Thu Oct 21 21:40:50 2021
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.7
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.8
  Git commit:       20.10.7-0ubuntu5~18.04.2
  Built:            Thu Oct 21 19:55:00 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.5.5-0ubuntu3~18.04.1
  GitCommit:
 runc:
  Version:          1.0.1-0ubuntu2~18.04.1
  GitCommit:
 docker-init:
  Version:          0.19.0
  GitCommit:
````

cpq, 2021-11-09:
The `-v` option needs a valid argument.
How do you invoke this command?

mailinglists35, 2021-11-09:
make linux

see the top of my snippet

btw, does that produce a mongoose_linux? that's all I want since I cannot find the binary on your website anymore.

cpq, 2021-11-09:
It is very difficult to figure out what you're doing.
Please include ALL INFORMATION - what command did you run, from what directory, on what OS.
Otherwise you're wasting a lot of time.

mailinglists35, 2021-11-09:
I still have an older copy but it fails to run:

````
root@srv:/usr/local/src/mongoose# ../mongoose_linux -g yes
2021-11-09 16:13:34  I main.c:114:cfg            Setting option [-g] to [yes]
2021-11-09 16:13:34  I sock.c:482:mg_listen      1 accepting on http://0.0.0.0:8000 (port 8000)
2021-11-09 16:13:34  E event.c:18:mg_error       2 TLS is not enabled
2021-11-09 16:13:34  I main.c:354:init_mongoose  Starting Mongoose v7.3.3, serving /usr/local/src/mongoose
2021-11-09 16:13:34  I main.c:355:init_mongoose  Local  URL: http://192.168.1.5:8000
2021-11-09 16:13:34  I main.c:357:init_mongoose  Global URL: https://38c09288dc1394.share.mongoose.ws
2021-11-09 16:13:35  E event.c:18:mg_error       4 TLS is not enabled
2021-11-09 16:13:35  E main.c:264:tun_fn         Tunnel closed
2021-11-09 16:13:35  E dns.c:182:dns_cb          stray DNS reply
2021-11-09 16:13:36  E event.c:18:mg_error       5 TLS is not enabled
2021-11-09 16:13:36  E main.c:264:tun_fn         Tunnel closed
2021-11-09 16:13:36  E dns.c:182:dns_cb          stray DNS reply
2021-11-09 16:13:36  E event.c:18:mg_error       6 TLS is not enabled
2021-11-09 16:13:36  E main.c:264:tun_fn         Tunnel closed
2021-11-09 16:13:37  E dns.c:182:dns_cb          stray DNS reply
2021-11-09 16:13:37  E event.c:18:mg_error       7 TLS is not enabled
2021-11-09 16:13:37  E main.c:264:tun_fn         Tunnel closed
2021-11-09 16:13:37  E dns.c:182:dns_cb          stray DNS reply
2021-11-09 16:13:38  E event.c:18:mg_error       8 TLS is not enabled
2021-11-09 16:13:38  E main.c:264:tun_fn         Tunnel closed
2021-11-09 16:13:38  E dns.c:182:dns_cb          stray DNS reply
2021-11-09 16:13:39  E event.c:18:mg_error       9 TLS is not enabled
2021-11-09 16:13:39  E main.c:264:tun_fn         Tunnel closed
````

cpq, 2021-11-10:
Please use https://github.com/cesanta/mongoose/tree/master/examples/http-server
Feel free to extend it with the functionality you need,

However it does not provide all functionality that was present in the binary - like, global sharing. That was closed source and we have no intention to open it.

closed 2021-11-10 by cpq