cesanta/mongoose #1403: errors on mg_tls_init when acting as client

rafalsk, 2021-11-15:
When the library as a client, often we're faced with errors during TLS session negotiation.
The errors look like the ones below:
```
tls hs: rc -1, err 5
tls hs: rc -1, err 6
```

Would you please advise where to find references in regards to these error codes?

Also, 

- how to enable for multiple cipher-suites? ex. for both TLS_AES_256_GCM_SHA384 and TLS_CHACHA20_POLY1305_SHA256?
- would it be possible to override TLS negotiation issues, such as due to an expired certificate on the other endpoint?

cpq, 2021-11-15:
@rafalsk thanks, but you've ignored issue template and your question is missing some important information about the environment. Also, what TLS lib / version you're using is missing.

rafalsk, 2021-11-15:
It's Windows and OpenSSL.

rafalsk, 2021-11-19:
Closing for now as more data is needed.

closed 2021-11-19 by rafalsk