cesanta/mongoose #1405: In the mg_connect_opt function, address is cleared

MaybenYi, 2021-11-17:
- My goal is:Obtain the correct IP address
- My actions were: In the mg_connect_opt function, perform debugging to discover faults
- My expectation was: Obtain the correct IP address
- The result I saw: In the mg_connect_opt function, address is cleared
- My question is:Why was the address cleared

test demo：
```
    DBG(("1:---%s", address));
    MG_COPY_COMMON_CONNECTION_OPTIONS(&add_sock_opts, &opts);
    DBG(("2:---%s", address));
    if ((nc = mg_create_connection(mgr, callback, add_sock_opts)) == NULL) {
        return NULL;
    }
    DBG(("3:---%s", address));
    if ((rc = mg_parse_address(address, &nc->sa, &proto, host, sizeof(host))) <
        0) {
        /* Address is malformed */
        MG_SET_PTRPTR(opts.error_string, "cannot parse address");
        mg_destroy_conn(nc, 1 /* destroy_if */);
        return NULL;
    }
```

output：
```
[mg:mg_connect_opt]1:---10.20.13.26:80
[mg:mg_connect_opt]2:---10.20.13.26:80
[mg:mg_connect_opt]3:---
```

# Environment

- mongoose version (e.g. 7.3): 6.7
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3):arm-none-eabi-gcc 8.5
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): RTOS（custom）
- Target hardware/board (e.g. STM32F411): STM32F429

MaybenYi, 2021-11-17:
I found the cause of the problem

In mg_http_common_url_parse()
```
  while (*url != '\0') {
    *addr = (char *) MG_REALLOC(*addr, addr_len + 6 /* space for port too. */);
    if (*addr == NULL) {
      DBG(("OOM"));
      return -1;
    }
```
Application addr memory, every expanded memory size is less than 32 bytes, in our rtos source code, is less than 32 bytes, can't again to apply for a new block, so lead to addr did not apply to the new memory address, behind other pointer at the time of application memory address, the addr directly to the memory address to take up.

We know that mongoose source code, according to the required byte size to apply for memory size, but we can not avoid our kernel memory management mechanism.

So the way I get around it is to allocate memory first and then assign

cpq, 2021-11-17:
Please use the recent version.

closed 2021-11-17 by cpq