cesanta/mongoose #1413: Problem with uri_to_path2  when serving directory using relative paths

rafalsk, 2021-11-25:
# Environment

- mongoose version (e.g. 7.3): 7.4
- Compiler/IDE (e.g. arm-none-eabi-gcc 9.3): Visual Studio 2019
- Target OS (e.g. FreeRTOS 10.3.1, Linux 5.8): Windows 10

We've just upgraded from 7.3 to 7.4 it was all good with 7.3

Seems like lots of code changes in regards to mg_http_serve_dir took place.

We provide a relative path to root_dir set in mg_http_serve_opts, relative to executable.

In 7.4 the server reports 404 with File Not found when attempting to access main root of localhost.

We've hunted the issue to function uri_to_path2, specifically the invocation of remove_double_dots, which seemingly destroys the path. All it takes to resolve the issue on our side is to comment out the invocation of remove_double_dots.

During the faulty situation debug data are as follows, when seen by uri_to_path2:
We use posix file system.

dir parameter: ".\\..\\..\\WebUI"
URL: "/"
start of HTTP message: ""GET / HTTP/1.1\r\nHost: localhost\r\nConnection: keep-alive\r\"

on line 1476 where remove_double_dots is invoked it is fed with ".\\..\\..\\WebUI/"

The function changes the above string to ".\\WebUI/"

Now the that is fed to _wstati64 which reports the path as invalid.

All we did was to comment out remove_double_dots() which did the trick.

cpq, 2021-11-25:
@rafalsk that function prevents directory traversal attacks, and it is essential.
Before it worked because `uri_to_path()` used `realpath()` function which turned your root_dir to absolute path.

Solution: use absolute path for your root directory, i.e. change

```c
    struct mg_http_serve_opts opts = {0};
    opts.root_dir = "..\..\WebUI";
    mg_http_serve_dir(c, ev_data, &opts);
```

to

```c
    char buf[MG_MAX_PATH];
    struct mg_http_serve_opts opts = {0};
    _fullpath("..\..\WebUI", buf, sizeof(buf));
    opts.root_dir = buf;
    mg_http_serve_dir(c, ev_data, &opts);
```

rafalsk, 2021-11-26:
@cpq thank you so much for your in-depth response and for the amazing piece of software you've created.

closed 2021-11-26 by rafalsk