cesanta/mongoose #1415: ESP32 crash when using mg_tls_init()

YanisHessini, 2021-11-29:
- My goal is: _Implement HTTPS._
- My actions were: 

1. _Adding another listener for HTTPS._
2. _Storing the strings for the certificates in `const char* cert, cert_key`._
3. _Initiating TLS with `mg_tls_init()` in the event handler._

- My expectation was: _Being able to connect via HTTPS to the ESP32 IP address and port._
- The result I saw: _**Crash after the TLS Init.**_

```
rst:0x1 (POWERON_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00     
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ETH Started
1970-01-01 00:00:00 2 Mongoose.cpp:3329:mg_liste 1 accepting on http://0.0.0.0:8000 (port 8000)
1970-01-01 00:00:00 2 Mongoose.cpp:3329:mg_liste 2 accepting on https://0.0.0.0:8443 (port 8443)
ETH Connected
ETH MAC: 08:3A:F2:3E:A2:BF, IPv4: 192.168.83.140, FULL_DUPLEX, 100Mbps      
Connection 0x3ffccad0 from 192.168.83.87:63830
TLS is_tls : 1
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d28bc  PS      : 0x00060c30  A0      : 0x800d28d4  A1      : 0x3ffb1da0
A2      : 0x3ffd168f  A3      : 0x3ffd168e  A4      : 0x40000000  A5      : 0x00000004
A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0x800d28bc  A9      : 0x3ffb1d80
A10     : 0x3c0fd9e6  A11     : 0xffffdeec  A12     : 0x008ab44e  A13     : 0xfffa03a3
A14     : 0xffffffff  A15     : 0x00000008  SAR     : 0x0000000c  EXCCAUSE: 0x00000003
EXCVADDR: 0x40000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x400d28bc:0x3ffb1da0 0x400d28d1:0x3ffb1dc0 0x4011d976:0x3ffb1de0 0x4011e79c:0x3ffb1e10 0x4011a778:0x3ffb1e30 0x4011a7a1:0x3ffb1e50 0x400d350d:0x3ffb1e70 0x400d422b:0x3ffb1e90 0x400d0f1e:0x3ffb1f90 0x400d8494:0x3ffb1fb0 0x400899ce:0x3ffb1fd0
  #0  0x400d28bc:0x3ffb1da0 in mg_random at lib/Mongoose/Mongoose.cpp:1542  
  #1  0x400d28d1:0x3ffb1dc0 in mbed_rng(void*, unsigned char*, unsigned int) at lib/Mongoose/Mongoose.cpp:1542
  #2  0x4011d976:0x3ffb1de0 in ssl_write_server_hello at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_srv.c:2495
  #3  0x4011e79c:0x3ffb1e10 in mbedtls_ssl_handshake_server_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_srv.c:4326
  #4  0x4011a778:0x3ffb1e30 in mbedtls_ssl_handshake_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:7165
  #5  0x4011a7a1:0x3ffb1e50 in mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:7165
  #6  0x400d350d:0x3ffb1e70 in mg_tls_handshake at lib/Mongoose/Mongoose.cpp:1542
  #7  0x400d422b:0x3ffb1e90 in mg_mgr_poll at lib/Mongoose/Mongoose.cpp:1542
  #8  0x400d0f1e:0x3ffb1f90 in loop() at src/main.cpp:225
  #9  0x400d8494:0x3ffb1fb0 in loopTask(void*) at C:/Users/ISAAN-Electronique/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23 
  #10 0x400899ce:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

# Environment

Mongoose version: _7.4_
- Compiler/IDE and SDK: ZZZ version _PlatformIO and Arduino Framework_
- Target hardware/board: _ESP32_
- Connectivity chip/module: _OLIMEX ESP32 GATEWAY_

After searching for a while in my code, I tried to search what was going on in the implementation of the `mg_random()` function, since it was mentioned in the backtrace.

```
#if MG_ENABLE_CUSTOM_RANDOM
#else
void mg_random(void *buf, size_t len) {
  bool done = false;
  unsigned char *p = (unsigned char *) buf;

#if MG_ARCH == MG_ARCH_ESP32
  while (len--) *p++ = (unsigned char) (esp_random() & 255);
#elif MG_ARCH == MG_ARCH_WIN32
#elif MG_ARCH_UNIX
  FILE *fp = fopen("/dev/urandom", "rb");
  if (fp != NULL) {
    if (fread(buf, 1, len, fp) == len) done = true;
    fclose(fp);
  }
#endif
  // Fallback to a pseudo random gen
  if (!done) {
    while (len--) *p++ = (unsigned char) (rand() & 255);
  }
}
#endif
```

In `mg_random()`, when using an ESP32 and therefore having `MG_ARCH == MG_ARCH_ESP32`, the variable `bool` is set to false at the beginning but **never to true**.

This leads to the Fallback instructions being executed, and it is likely the cause of my device's crash. When I comment the fallback, the HTTPS code works as intended.

cpq, 2021-12-01:
Did you try the latest HEAD ?

YanisHessini, 2021-12-01:
It seems that I wasn't using the latest one. The code available on this repo has fixed this by adding : `done = true;`, in the .c file.

Thank you for pointing this out. I will pull the files again.

cpq, 2021-12-09:
Thank you. Closing this.

closed 2021-12-09 by cpq