cesanta/mongoose #1416: How to make http server to support ipv6 ?

guangjung, 2021-11-30:
- My goal is: To support ipv6 (listen any eth)
- My actions were: 
I modify http-restful-server example source code 
step1. modify Makefile by set -DMG_ENABLE_IPV6=1

all: $(PROG)
        $(DEBUGGER) ./$(PROG) $(ARGS)

$(PROG): main.c
        $(CC) ../../mongoose.c -I../.. -W -Wall -DMG_ENABLE_LINES=1 **-DMG_ENABLE_IPV6=1** $(CFLAGS) -o $(PROG) main.c
....
step2: modify main.c

**static const char *s_http_addr = "http://[2001:db8:0:f101::8]:8000";**    // HTTP port
static const char *s_https_addr = "https://0.0.0.0:8443";  // HTTPS port

step3：make  ， print error
# make
./example 
2021-11-30 08:31:15 1 sock.c:200:mg_open_listene Failed to listen on http://[2001:db8:0:f101::8]:8000, errno 22
2021-11-30 08:31:15 1 sock.c:489:mg_listen       Failed: http://[2001:db8:0:f101::8]:8000, errno 22

To use **static const char *s_http_addr = "http://[::]:8000";** , error is the same

# Environment
Ubuntu 16.04

- mongoose version: lastest version
- Compiler/IDE and SDK:  gcc

guangjung, 2021-11-30:
Sorry, Maybe my linux server environmental problems. NOW run ok but I NOT found root cause. This is a strange phenomenon.

If Set s_http_addr = "http://[::]:8000, ipv4 and ipv6 are both supported, but ONLY found tcp6 listen in Ubuntu. why?
$ netstat -plant | grep 8000
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp6       0      0 :::8000                 :::*                    LISTEN      -               
guangjung@linux:~$

cpq, 2021-12-09:
Please don't file issues that are not issues.

closed 2021-12-09 by cpq