cesanta/mongoose #1420: [Question] Retry TLS handshake after it failed for same mg_connection without killing it

rafalsk, 2021-12-03:
We've spent quite a lot of time on this; to no avail so far.

With Mongoose acting as client, In some cases remote web-servers do not properly handle SNI requests.

In such a case we would like to retry TLS handshake (OpenSSL 1.1.1) using same mg_connection without killing it and specifying another (ex. empty hostname) during TLS handshake (we're well aware of opts->srvname option given to SSL_set_tlsext_host_name().

We can see mg_tls_handshake is called from connect_conn() and mg_tls_handshake_retry() the latter directly from mg_mgr_poll,
It usually gets called twice from both of the above functions and succeeds the second time (we didn't really get to understand what that is so, so far), or why both invocations needs to happen.

We've been implementing sort of a proxy supporting persistent connection thus we're trying to get around the asynchronous nature of Mongoose  (events polling) and would be most happy with a couple of synchronous TLS handhake9s) performed right after connect, while the client awaits a response.

Anyway. You might get an idea of what we've tried by looking at the snippet below:
`void mg_tls_handshake_retry(struct mg_connection *c, int tries) {
  struct mg_tls *tls = (struct mg_tls *) c->tls;
  int rc;

  if (c->is_client && tries >= 2)
  {
      if(tries==2)
       {
          //SSL_clear(tls->ssl)
          mg_tls_free(c);
          CCORSConnCtx *ctx = reinterpret_cast<CCORSConnCtx *>(c->fn_data); 
          mg_tls_opts *opts = ctx->getClientTLSOpts();
          opts->srvname = mg_str("");
          c->is_resolving=1;
    
          mg_tls_init(c, opts);
          tls = (struct mg_tls *) c->tls;
        
       }

  }

  SSL_set_fd(tls->ssl, (int) (size_t) c->fd);
  rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);

  if (rc == -1 && c->is_client && tries<6) {
    mg_tls_handshake_retry(c,++tries);
    tls = (struct mg_tls *) c->tls;
  }
  if (!c->is_tls_hs)
   {
      rc=1;//it must have succeeded during recursion
   }

  if (rc == 1) {
    LOG(LL_DEBUG, ("%lu success", c->id));
    c->is_tls_hs = 0;
  } else {
    int code = mg_tls_err(tls, rc);
    if (code != 0) mg_error(c, "tls hs: rc %d, err %d", rc, code);
  }
}

The above modified `mg_tls_handshake manages to connect synchronously (without waiting for another poll) when remote server support SNI just fine i.e. provides a proper response. But when it does not (the recursion gets above depth equal to 2 (the tries variable), that happens only in such cases, then we free the entire TLS related stuff associated with the connection and create it anew through mg_tls_init().

For some reason the TLS handshake does not work. Had we specified a proper and/or supported SNI hostname, it would have worked just fine. 

**We just do not seem to be able to fully resurrect the TLS handshake with ANOTHER SNI hostname. That's what we're after.**

cpq, 2021-12-15:
Thank you for the excellent description.
That seems subtle and deep.

What if Mongoose fetched more TLS error data in handshake, e.g. TLS error number, and fired a error which can be parsed by the error handler? E.g. the error message look like `TLS error 40: ...`

Then the Mongoose event handler can process it:
```c
static void fn(struct mg_connection *c, int ev, ...) {
  switch (ev) {
    case MG_EV_ERROR:
      if (strncmp((char *) ev_data, "TLS error 40", 12) == 0) {
         // Re-initialize the connection. Optionally, keep reconnection cound in c->label
        mg_iobuf_free(&c->send);
        mg_iobuf_free(&c->recv);
         ... clear other flags ..
         c->is_closing = 0;  // Clear error
         struct mg_tls_opts opts = {...};
         mg_tls_init(c, &opts);
      }
    break;
  }
}
```
Not sure that is going to work.

rafalsk, 2021-12-17:
More data regarding TLS-handshake event passed to the event handler would be welcome!

It would surely enable for high-level inferring that some further customized processing might be needed.

I know you support more than 1 TLS library so adding customized flags/enums for each family of events and translating across these would be an overfill. 

To us, more data within a string is good enough.

Even if one feels up for some customized lower-level processing at the level of TLS library and/or looks at the TLS structures, having the possibility to retrieve hints from the event seems rational.

 Add mg_tls_init_modify_conf() #1426 ~ Great!

2. Amazing.

Closing this for now, I would allow myself to reopen once we get more data.
I shall also give you head-ups about our further results in any case.

Thank you so much.

closed 2021-12-17 by rafalsk

cpq, 2021-12-17:
@rafalsk if you have a diff that generates suitable handshake error string, please share - we'll integrate it.