cesanta/mongoose #1429: webSocket automatic disconnect with log  mongoose.c:403:mg_error    1 unknown WS op 15

ShanXueQinSi, 2021-12-18:
- My goal is: Upload logs in the production environment(WS server) to the developer's computer(WS client) in real time
- My expectation was: webSocket  automatic disconnect with log in client  
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 15
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 15
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 15
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 15
                            2021-12-18 02:42:59 1 mongoose.c:4689:mg_ws_cb   1 Got WS CLOSE
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 3
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 3
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 3
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 3
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 3
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 3
                            2021-12-18 02:42:59 1 mongoose.c:4689:mg_ws_cb   1 Got WS CLOSE
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 3
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 3
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 5
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 5
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 3
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 3
                            2021-12-18 02:42:59 1 mongoose.c:403:mg_error    1 unknown WS op 5
                            2021-12-18 02:42:59 1 main.c:16:fn               0x3 unknown WS op 5
# Environment
- mongoose version: 7.4
- Compiler/IDE and SDK: gcc
- Target hardware/board: server arm64 / client ubuntu 18.04
-client code : https://github.com/cesanta/mongoose/tree/7.4/examples/websocket-client
-server code :  modify to sending informations non-stop with sample https://github.com/cesanta/mongoose/tree/7.4/examples/websocket-server

cpq, 2021-12-18:
No idea what you're doing and what exactly code you're using.
Please learn how to report issues adequately.

closed 2021-12-18 by cpq

ryuukk, 2021-12-22:
@ShanXueQinSi you are using old version, try the latest 7.5 to make sure you don't miss any fixes

also a little sample code to reproduce the issue will make it easier to find what is the issue exctly