cesanta/mongoose #1430: MG_EV_HTTP_MSG event does not fire in http-revere-proxy example

jcorporation, 2021-12-20:
- My goal is: Implement a reverse proxy that manipulates the retrieved data
- My actions were: used the http-reverse-proxy example and added the MG_EV_HTTP_MSG event
- My expectation was: MG_EV_HTTP_MSG event is fired as documented
- The result I saw: MG_EV_HTTP_MSG event is NOT fired
- My question is: Should this event fire? Your answer in #1399 says "yes"

# Environment

- mongoose version: current master
- Compiler/IDE and SDK: n/a
- Target hardware/board: n/a
- Connectivity chip/module: n/a

I replaced the fn2 function of the http-reverse-proxy example with this snippet:
```
 static void fn2(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_READ && fn_data != NULL) {
    printf("MG_EV_READ\n");
  }
  if (ev == MG_EV_HTTP_MSG) {
    printf("MG_EV_HTTP_MSG\n");
    struct mg_http_message *hm = (struct mg_http_message *) ev_data;
    mg_http_reply((struct mg_connection *) fn_data, 200,
      "Content-Typ: text/html", "%.*s", hm->body.len, hm->body.ptr);
   }
 }

```

Log
```
2021-12-20 14:24:51 3 mongoose.c:3352:mg_listen  1 accepting on http://localhost:8080 (port 8080)
2021-12-20 14:24:54 3 mongoose.c:3255:accept_con 2 accepted 127.0.0.1:58034
2021-12-20 14:24:54 3 mongoose.c:3223:mg_connect 3 -> http://cesanta.com
2021-12-20 14:24:54 3 mongoose.c:3223:mg_connect 4 -> udp://8.8.8.8:53
2021-12-20 14:24:54 3 main.c:31:forward_request  FORWARDING: GET /
2021-12-20 14:24:54 3 mongoose.c:272:dns_cb      3 cesanta.com resolved to 148.251.54.236
MG_EV_READ
MG_EV_READ
MG_EV_READ
2021-12-20 14:25:28 3 mongoose.c:3082:read_conn  2   00100000000010       0 -1/128 err 115
2021-12-20 14:25:28 3 mongoose.c:3144:close_conn 2 closed
2021-12-20 14:25:29 3 mongoose.c:3144:close_conn 3 closed

```

`curl http://127.0.0.1:8080` does not return after sending the request.

cpq, 2021-12-20:
The event should not fire.

The reason is that the connection to the upstream server is not HTTP. It is plain TCP.

jcorporation, 2021-12-20:
Thanks for your fast answer. Is there a simple way to build a http aware reverse proxy with mongoose?

cpq, 2021-12-20:
There is. Instead of plain TCP, use HTTP.
Why? Is that for rewriting headers or something like that?

jcorporation, 2021-12-20:
Yes, I want to validate and rewrite the response. Usage of `mg_http_connect` instead of `mg_connect` is the right way?

cpq, 2021-12-20:
Yes, correct. The logic should be very similar - just use `mg_http_connect` and instead of `MG_EV_READ`, catch `MG_EV_HTTP_MSG`

jcorporation, 2021-12-20:
I searched and searched for the error and wholly overseen the mg_connect call... now its clear. Thank you very much

closed 2021-12-20 by jcorporation