cesanta/mongoose #1432: SSL: load certifications from memory

ryuukk, 2021-12-21:
- **My goal is**:  to load certificates in system without file access
- **My actions were **: i looked at the code and seen only usage of SSL api that loads files from disk
- **My expectation was **: i was expecting to be able to load the certificates from memory considered the scope of the project
- **The result I saw **: i couldn't
-  **My question is **: Would it be possible to add an API that load certificates from memory instead of disk?

# Environment

- mongoose version: _7.5_
- Compiler/IDE and SDK: ZZZ version _X.Y_
- Target hardware/board: _?_
- Connectivity chip/module: _?_

cpq, 2021-12-21:
Please specify your environment and what SSL library you're using, or this will be closed with no further attention.

ryuukk, 2021-12-21:
@cpq sorry i should have been more precise, i'll close the issue for now and instead open a discussion thread, i am currently working on a PR to add the feature have in mind, that'll be easier to explain for me!

closed 2021-12-21 by ryuukk

ptsolakos, 2021-12-21:
@ryuukk : The same API can be used to load certificates from memory.
Have a look in the detailed documentation of `struct mg_tls_opts` [here](https://mongoose.ws/documentation/#struct-mg_tls_opts):
`const char *cert` : Can be a filename, or a string. In case of string, the certificate in PEM encoding from memory can be given.

@cpq: Unfortunately there is no way to pass certificates in DER encoding from memory with the current API. That is indeed a shortcoming.

ryuukk, 2021-12-21:
@ptsolakos are you sure? i couldn't in my test, i'm not very experienced with openssl unfortunately:

```
139815341569024:error:02001002:system library:fopen:No such file or directory:crypto/bio/bss_file.c:69:fopen('-----BEGIN CERTIFICATE-----
(...)
-----END CERTIFICATE-----
','r')
139815341569024:error:2006D080:BIO routines:BIO_new_file:no such file:crypto/bio/bss_file.c:76:
139815341569024:error:0B084002:x509 certificate routines:X509_load_cert_crl_file:system lib:crypto/x509/by_file.c:199:
2021-12-21 20:38:37 1 mongoose.c:401:mg_error    1 parse(-----BEGIN CERTIFICATE-----
(...)
-----END CERTIFICATE-----
): err 5
```

ryuukk, 2021-12-21:
Maybe i am misunderstanding something? 

If someone have an example to follow that would be nice, because to be honest i am a little bug confused right now..

The use case:

A secure websocket server

A websocket client connecting to that server

I have been following this: https://mongoose.ws/tutorials/tls/

But doesn't cover the loading keys as string, hence the error i am currently having

Thanks!

ptsolakos, 2021-12-22:
@ryuukk I am sorry for the confusion. Your code snippet is correct, however I found out now that this feature of mongoose is currently only supported when you are using MbedTLS and not OpenSSL.

ryuukk, 2021-12-22:
> @ryuukk I am sorry for the confusion. Your code snippet is correct, however I found out now that this feature of mongoose is currently only supported when you are using MbedTLS and not OpenSSL.

i'll try to come up with a PR to allow it with OpenSSL this week end

mongoose is an awesome library, i have to contribute back something!

inaryart, 2021-12-24:
> i'll try to come up with a PR to allow it with OpenSSL this week end
>
> mongoose is an awesome library, i have to contribute back something!

waiting for  this feature

shu7734, 2023-05-24:
feature is worked now?