cesanta/mongoose #1435: Can't compile 7.5 with MinGW since gmtime_R and localtime_r definitions

marcelwirtz, 2021-12-23:
I tried to upgrade from Mongoose Version 7.3 to 7.5 but every time i try to compile i get the compile error below.

C:\PROJECTS\CPP\libmgwebserver\src\server\mongoose.h:454: In file included from ..\src\server\mongoose.h:454,
In file included from ..\src\server\mongoose.h:454,
                 from ..\src\server\mongoose.c:18:
C:/Qt/Tools/mingw810_32/i686-w64-mingw32/include/time.h:312:34: note: previous definition of 'gmtime_r' was here
 __forceinline struct tm *__cdecl gmtime_r(const time_t *_Time, struct tm *_Tm) {

Is there sth. i can do? 7.3 worked like a charm but used normal gmtime and localtime which should be thread safe already on Windows if i got this right.

I wish you happy holidays and a happy new year.

Kind regards,
Marcel

cpq, 2021-12-23:
The `make mingw` target runs a unit test using mingw compiler, and it works fine. Which mingw version do you use?
I think I'll ban people who ignore issue questions, which ask about the compiler. Why did you ignore it and force me to spend my time on this round-trip?

marcelwirtz, 2021-12-24:
To be honest it wasn't on purpose to ignore it. I copy and pasted the error message which also deleted the questions, got distracted by a phone call and forgot about it because i had to do other stuff first. So sorry about that!

- mongoose version: 7.5 (with OpenSSL 1.1.1h)
- Compiler/IDE and SDK: MinGW / gcc version 8.1.0 (i686-posix-dwarf-rev0, Built by MinGW-W64 project) which ships with the Qt Framework
- Target hardware/board: i686-w64-mingw32

Hope thats all things you need to know besides the error message above.

What i already tested yesterday: 

if i put 

```
#ifdef _POSIX_THREAD_SAFE_FUNCTIONS
    #warning "With POSIX"
#else
    #warning "Without POSIX"
#endif

```
right before the implementations in mongoose.h, to see if the POSIX functions from time.h to be used. I am getting the output in the screenshot which i can't explain why the first time POSIX isn't used and then it is a few times.

[image]

Kind regards and again sorry for the inconvenience,
Marcel

cpq, 2021-12-24:
In Mongoose's `make mingw` the GCC 9.3.0 is used:

```bash
$ docker run --rm -e Tmp=. -e WINEDEBUG=-all -v /home/cpq/src/mongoose:/home/cpq/src/mongoose -w /home/cpq/src/mongoose mdashnet/mingw i686-w64-mingw32-gcc -v
```

```bash
Target: i686-w64-mingw32
...
gcc version 9.3.0 (GCC) 
```

The `make mingw`triggers the following command:
```sh
$ make mingw
docker run --rm -e Tmp=. -e WINEDEBUG=-all -v /home/cpq/src/mongoose:/home/cpq/src/mongoose -w /home/cpq/src/mongoose mdashnet/mingw i686-w64-mingw32-gcc mongoose.c test/unit_test.c test/packed_fs.c -W -Wall -Werror -I. -DMG_MAX_HTTP_HEADERS=7 -DMG_ENABLE_LINES -DMG_ENABLE_PACKED_FS=1 -lwsock32 -o test.exe
```

So my guess is that on your system, some build options enable `_r` functions, hence you get a conflict.
The right way to fix that would be to guard the appropriate block in src/arch_win32.h. If you can find which preprocessor definition switches on those `_r` functions, that'll be great.

nnieddu, 2021-12-24:
Hello @marcelwirtz,

Just try :
`gcc ../../mongoose.c main.c -I../.. -W -Wall -DMG_ENABLE_LINES=1 -lws2_32 -o mongoose.exe`
(without the "-D_POSIX_C_SOURCE=200000L" compile flag).

You may juste have some warnings but not a problem.
[image]

Tested today with last mongoose(7.5) release and last WinLibs (MinGW-w64 9.0.0).

nnieddu, 2021-12-24:
Hello @marcelwirtz,

Just try compiling with clang and without the "-D_POSIX_C_SOURCE=200000" :

`clang ../../mongoose.c main.c -I../.. -W -Wall -DMG_ENABLE_LINES=1 -lws2_32 -o mongoose.exe`

Tested today with last mongoose(7.5) release and [GCC 11.2.0 + LLVM/Clang/LLD/LLDB 13.0.0 + MinGW-w64 9.0.0 ](https://winlibs.com/), everything seems working well.
(just download winlib archive extract where you want and add the "mingw64/bin/" to your PATH env variable).

marcelwirtz, 2021-12-27:
Hey guys,
thanks for helping out. Since I can't get rid of POSIX definitions as they come from the Qt framework that we use and the functions come from time.h where they are enabled with _POSIX_THREAD_SAFE_FUNCTIONS define, i guarded the definitions in mongoose.h with this, whats seems to work for me.

```
#ifndef _POSIX_THREAD_SAFE_FUNCTIONS
// https://lgtm.com/rules/2154840805/ -gmtime, localtime, ctime and asctime
static __inline struct tm *gmtime_r(time_t *t, struct tm *tm) {
  (void) tm;
  return gmtime(t);
}

static __inline struct tm *localtime_r(time_t *t, struct tm *tm) {
  (void) tm;
  return localtime(t);
}
#endif
```
I don't know if it's possible to do that by default but anyway thanks for pointing that out.

Kind regards and merry christmas to you,
Marcel

closed 2021-12-27 by marcelwirtz

bing2008, 2022-01-18:
Hey @marcelwirtz ,this is my bad. I was just test the version in my env (mingw32) , that's  make the issue. 
see: https://github.com/cesanta/mongoose/discussions/1408#discussioncomment-1812139