cesanta/mongoose #1437: compilation issue.

vrkansagara, 2021-12-26:
I am not sure about this error but make should provide understandable message that why it's failed.

~~~bash
$ make
2021-12-26 13:06:29 3 sock.c:232:read_conn       3   01000100000010   10627 -30848/3709 err 11
2021-12-26 13:06:29 3 sock.c:294:close_conn      3 closed
2021-12-26 13:06:29 3 sock.c:373:mg_connect      4 -> http://ipv6.google.com
2021-12-26 13:06:29 3 sock.c:373:mg_connect      5 -> udp://[2001:4860:4860::8888]:53
2021-12-26 13:06:32 1 event.c:21:mg_error        4 DNS timeout
2021-12-26 13:06:32 3 sock.c:294:close_conn      4 closed
FAILURE test/unit_test.c:788: ok == 200
make: *** [Makefile:65: test] Error 1

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

$ unam -a
Linux  5.4.0-91-generic #102-Ubuntu SMP Fri Nov 5 16:31:28 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

~~~

cpq, 2021-12-27:
Can you `ping6 ipv6.google.com` from your terminal?

vrkansagara, 2021-12-30:
```bash
ping6: ipv6.google.com: Name or service not known
```

cpq, 2021-12-30:
So, this is an issue with IPv6 setup on your workstation.
You should see something like this:

```sh
 ping6 ipv6.google.com
PING ipv6.google.com(di-in-f102.1e100.net (2a00:1450:400b:c01::66)) 56 data bytes
64 bytes from di-in-f102.1e100.net (2a00:1450:400b:c01::66): icmp_seq=1 ttl=110 time=14.7 ms
64 bytes from di-in-f102.1e100.net (2a00:1450:400b:c01::66): icmp_seq=2 ttl=110 time=14.9 ms
```

Next time, please be more careful and do not report an issue which is not an issue.

closed 2021-12-30 by cpq

vrkansagara, 2021-12-31:
@cpq  This is something related to hardware where I am compiling so I feel that code has some kind of check which identify that ipv6 is natively enable and used by os and hardware or not.

I will be more careful about reporting issue @cpq, Thanks for that word.