cesanta/mongoose #1440: Mongoose 7.X does not receive data on UDP socket connected to multicast group.

mill1000, 2022-01-01:
# Environment
- mongoose version: 7.5
- Compiler/IDE and SDK: esp-idf v4.2.2
- Target hardware/board: ESP32-D0WD-V3

# Description
I am trying to port a section of code from Mongoose 6.18 to Mongoose 7.5. The code performs a SSDP search which consists of sending UDP packets to a multicast group `udp://239.255.255.250:1900` and waiting for responses. The other devices respond by sending UDP data back to the source IP and port of the searching device.

e.g.
`<my-ip>:<random port A> -> 239.255.255.250:1900`
`<other-device>:<random port B> -> <my-ip>:<random port A>`

Mongoose 7 does not receive the response and Wireshark captures show that the port is unreachable.

### Mongoose 6.18
This is the expected behavior.
Log output:
```
I (807) WiFi: Connected.
I (855) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1570) esp_netif_handlers: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (1570) WiFi: Assigned IP: 192.168.1.105
I (5969) Main: Sending M-SEARCH.
I (5970) Main: Got event: 0
I (5970) Main: Got event: 2
I (5971) Main: Got event: 4
I (5979) Main: Got event: 0
I (5980) Main: Got data!
```
and we can see that a device did respond.
[image]

### Mongoose 7.5
Mongoose 7 never receives data, it just continues to poll indefinitely
Log output:
```
I (882) WiFi: Connected.
I (937) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1561) esp_netif_handlers: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (1562) WiFi: Assigned IP: 192.168.1.105
I (7851) Main: Sending M-SEARCH.
I (7851) Main: Got event: 1
I (7851) Main: Got event: 3
I (7852) Main: Got event: 4
I (7855) Main: Got event: 2
I (8858) Main: Got event: 2
I (9858) Main: Got event: 2
```
and we see the port was deemed unreachable.
[image]

This attached code demonstrates this problem and can be compiled for Mongoose 7.5 and Mongoose 6.18 by using the `MONGOOSE_7` define. Of course this example assumes there is a device capable of responding on the network.

```c++
#include "esp_log.h"
#include "mongoose.h"

#define TAG "Main"

void send_search_request(struct mg_mgr* mgr)
{
  // SSDP search request to send
  static const char* search_request =\
    "M-SEARCH * HTTP/1.1\r\n"\
    "HOST: 239.255.255.250:1900\r\n"\
    "MAN: \"ssdp:discover\"\r\n"\
    "ST: urn:schemas-upnp-org:device:MediaRenderer:1\r\n"\
    "MX: 5\r\n"\
    "\r\n";

  ESP_LOGI(TAG, "Sending M-SEARCH.");

  // Create a outbound UDP socket
  struct mg_connection* search = mg_connect(mgr, "udp://239.255.255.250:1900", [](struct mg_connection* c, int ev, void* ev_data, void* fn_data)
  {
#ifdef MONGOOSE_7
    if (ev == MG_EV_READ)
#else
    if (ev == MG_EV_RECV)
#endif   
      ESP_LOGI(TAG, "Got data!");
    else
      ESP_LOGI(TAG, "Got event: %d", ev);
  }, nullptr);

  // Send search request
  mg_printf(search, search_request);
}

void run_mongoose()
{
  // Create and init a Mongoose manager
  struct mg_mgr manager;
#ifdef MONGOOSE_7
  mg_mgr_init(&manager);
#else
  mg_mgr_init(&manager, nullptr);
#endif

  // Setup a timer to perform the search after a delay
#ifdef MONGOOSE_7
  struct mg_timer timer;
  mg_timer_init(&timer, 5000, 0, [](void* fn_data){
    send_search_request((struct mg_mgr*) fn_data);
  }, &manager);
#else
  struct mg_connection* c = mg_bind(&manager, "80", [](struct mg_connection* c, int ev, void* ev_data, void* fn_data){
    if (ev == MG_EV_TIMER)
      send_search_request(c->mgr);
  }, nullptr);
  
  mg_set_timer(c, mg_time() + 5);
#endif
  
  // Loop waiting for events
  while(true)
    mg_mgr_poll(&manager, 1000);

  // Free the manager if we ever exit
  mg_mgr_free(&manager);
}
```

cpq, 2022-01-02:
Thank you @mill1000 
Can you increase the log level to e.g. 3 or even 4, and capture+share logs when search request gets sent, please?
To increase the log level, `mg_log_set("4");`

mill1000, 2022-01-02:
Sure thing. Here's the log output at "4".

```
I (921) WiFi: Connected.
I (929) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1562) esp_netif_handlers: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (1562) WiFi: Assigned IP: 192.168.1.105
I (7890) Main: Sending M-SEARCH.
1970-01-01 00:00:07 3 mongoose.c:3223:mg_connect 1 -> udp://239.255.255.250:1900
I (7890) Main: Got event: 1
I (7893) Main: Got event: 3
I (7897) Main: Got event: 4
I (7901) Main: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (8909) Main: Got event: 2
1970-01-01 00:00:08 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (9910) Main: Got event: 2
1970-01-01 00:00:09 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (10911) Main: Got event: 2
1970-01-01 00:00:10 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (11912) Main: Got event: 2
1970-01-01 00:00:11 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (12913) Main: Got event: 2
1970-01-01 00:00:11 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (13914) Main: Got event: 2
1970-01-01 00:00:12 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
I (14915) Main: Got event: 2
1970-01-01 00:00:13 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
```

I just noticed that we also don't see a `MG_EV_WRITE` even though Wireshark shows data going out.

cpq, 2022-01-05:
@mill1000 Thank you!

Mongoose did not log data sent via UDP, and did not send MG_EV_WRITE event.
Pushed https://github.com/cesanta/mongoose/commit/6a82cbee71cc22805776efe9dd1e2ef313581534 to fix that.  Also pushed a change that shows IP/PORT pair in the log, and added an SSDP example based on your code: https://github.com/cesanta/mongoose/tree/master/examples/udp-ssdp-search

Could you:
1. Pull the current code
2. Set c->is_hexdumping on the UDP connection (just like in the example)
3. 2. Run and make sure that MG_EV_WRITE is received, and the message is shown in the log
4. From the log, get IP/PORT and send a UDP message to the ESP board manually using `nc` utility

mill1000, 2022-01-06:
Cool. Updated to commit fc3c08e411eb750766794ee83d3f605f7e369b11

Here's the log now. There is a MG_EV_WRITE event now, but still no data coming back in. I used the following command `nc -u 192.168.1.105 60301` and sent the following "Hellllllooooo". I noticed that the hexdump gave the source as 0.0.0.0:60301, could the socket not be bound to an IP and thus it's ignoring data coming back?
```
I (800) WiFi: Connected.
I (827) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1561) esp_netif_handlers: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (1561) WiFi: Assigned IP: 192.168.1.105
I (7766) Main: Sending M-SEARCH.
1970-01-01 00:00:06 3 mongoose.c:3234:mg_connect 1 -> udp://239.255.255.250:1900
I (7766) Main: Got event: 1
I (7769) Main: Got event: 3
I (7773) Main: Got event: 4
1970-01-01 00:00:06 4 mongoose.c:2967:iolog      1   01000001010000 0:0 129 err 0
1970-01-01 00:00:07 2 mongoose.c:2985:iolog      
-- 1 0.0.0.0:60301 -> 239.255.255.250:1900  129
0000  4d 2d 53 45 41 52 43 48 20 2a 20 48 54 54 50 2f  M-SEARCH * HTTP/
0010  31 2e 31 0d 0a 48 4f 53 54 3a 20 32 33 39 2e 32  1.1..HOST: 239.2
0020  35 35 2e 32 35 35 2e 32 35 30 3a 31 39 30 30 0d  55.255.250:1900.
0030  0a 4d 41 4e 3a 20 22 73 73 64 70 3a 64 69 73 63  .MAN: "ssdp:disc
0040  6f 76 65 72 22 0d 0a 53 54 3a 20 75 72 6e 3a 73  over"..ST: urn:s
0050  63 68 65 6d 61 73 2d 75 70 6e 70 2d 6f 72 67 3a  chemas-upnp-org:
0060  64 65 76 69 63 65 3a 4d 65 64 69 61 52 65 6e 64  device:MediaRend
0070  65 72 65 72 3a 31 0d 0a 4d 58 3a 20 35 0d 0a 0d  erer:1..MX: 5...
0080  0a                                               .

I (7847) Main: Got event: 7
I (7851) Main: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (8860) Main: Got event: 2
1970-01-01 00:00:08 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (9861) Main: Got event: 2
1970-01-01 00:00:08 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (10862) Main: Got event: 2
1970-01-01 00:00:09 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
....
```

cpq, 2022-01-06:
The `0.0.0.0` is a catch-all address (INADDR_ANY), so it should work - but apparently it does not seem to receive data..
Could you also start a listening connection, e.g. `mg_listen(&mgr, "udp://0.0.0.0:1234", fn, NULL);` and send data to port 1234?

mill1000, 2022-01-06:
Creating a listening connection works just fine. I added the following to my above test code, and sent a string via nc.

```c++
mg_listen(&manager, "udp://0.0.0.0:1234", [](struct mg_connection* c, int ev, void* ev_data, void* fn_data)
  {
    if (ev == MG_EV_OPEN) 
      c->is_hexdumping = 1;

    if (ev == MG_EV_READ)
      ESP_LOGI(TAG, "Listner: Got data!");
    else
      ESP_LOGI(TAG, "Listner: Got event: %d", ev);
  }, nullptr);

  struct mg_timer timer;
  mg_timer_init(&timer, 5000, 0, [](void* fn_data){
    send_search_request((struct mg_mgr*) fn_data);
  }, &manager);
```

And the log
```
I (808) WiFi: Connected.
I (813) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1563) esp_netif_handlers: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (1563) WiFi: Assigned IP: 192.168.1.105
I (1773) Main: Listner: Got event: 2
1970-01-01 00:00:01 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (2774) Main: Listner: Got event: 2
1970-01-01 00:00:02 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (3775) Main: Listner: Got event: 2
1970-01-01 00:00:03 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (4776) Main: Listner: Got event: 2
1970-01-01 00:00:04 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (5777) Main: Listner: Got event: 2
1970-01-01 00:00:05 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (6778) Main: Listner: Got event: 2
1970-01-01 00:00:06 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (7779) Main: Sending M-SEARCH.
1970-01-01 00:00:06 3 mongoose.c:3234:mg_connect 2 -> udp://239.255.255.250:1900
I (7779) Main: Got event: 1
I (7782) Main: Got event: 3
I (7786) Main: Got event: 4
1970-01-01 00:00:06 4 mongoose.c:2967:iolog      2   01000001010000 0:0 129 err 88
1970-01-01 00:00:06 2 mongoose.c:2985:iolog      
-- 2 0.0.0.0:61237 -> 239.255.255.250:1900  129
0000  4d 2d 53 45 41 52 43 48 20 2a 20 48 54 54 50 2f  M-SEARCH * HTTP/
0010  31 2e 31 0d 0a 48 4f 53 54 3a 20 32 33 39 2e 32  1.1..HOST: 239.2
0020  35 35 2e 32 35 35 2e 32 35 30 3a 31 39 30 30 0d  55.255.250:1900.
0030  0a 4d 41 4e 3a 20 22 73 73 64 70 3a 64 69 73 63  .MAN: "ssdp:disc
0040  6f 76 65 72 22 0d 0a 53 54 3a 20 75 72 6e 3a 73  over"..ST: urn:s
0050  63 68 65 6d 61 73 2d 75 70 6e 70 2d 6f 72 67 3a  chemas-upnp-org:
0060  64 65 76 69 63 65 3a 4d 65 64 69 61 52 65 6e 64  device:MediaRend
0070  65 72 65 72 3a 31 0d 0a 4d 58 3a 20 35 0d 0a 0d  erer:1..MX: 5...
0080  0a                                               .

I (7860) Main: Got event: 7
I (7864) Main: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3451:mg_mgr_pol 2 -- tchrc
I (7873) Main: Listner: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
I (8503) Main: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3451:mg_mgr_pol 2 -- tchrc
I (8503) Main: Listner: Got event: 2
1970-01-01 00:00:07 4 mongoose.c:3451:mg_mgr_pol 1 r- tchrc
1970-01-01 00:00:07 4 mongoose.c:2967:iolog      1   10000001010010 0:0 11 err 88
1970-01-01 00:00:07 2 mongoose.c:2985:iolog      
-- 1 0.0.0.0:1234 <- 192.168.1.193:54566  11
0000  4c 69 73 74 65 6e 6e 6e 6e 6e 0a                 Listennnnn.

I (8532) Main: Listner: Got data!
I (9536) Main: Got event: 2
1970-01-01 00:00:08 4 mongoose.c:3451:mg_mgr_pol 2 -- tchrc
I (9536) Main: Listner: Got event: 2
1970-01-01 00:00:08 4 mongoose.c:3451:mg_mgr_pol 1 -- tchrc
```

cpq, 2022-01-06:
It looks like the issue is with `connect()` syscall.
If a `connect()` is called on a UDP socket, it is apparently considered to be an outbound connection and the IP stack refuses incoming data.
Pushed a change to skip `connect()` on a UDP connection. Please pull and retry.

mill1000, 2022-01-07:
Looks to be working now. I was able to send data via nc and also saw some search responses coming in too.

gegzy, 2022-01-07:
Not sure if this is me, but I think the new change has an adverse affect of changing the outgoing peer address to the last address it received data from, meaning the search request no longer directs towards the multicast address...

Using the udp_ssdp-search example (with the hex output removed and additional LOG at the end for clarity), we can see that 192.168.1.1 replies to the search:

```
./example 
2022-01-07 08:37:35 2 main.c:14:fn               [ev=1] Peer is now: 0.0.0.0
2022-01-07 08:37:35 2 main.c:14:fn               [ev=3] Peer is now: 239.255.255.250
2022-01-07 08:37:35 2 main.c:14:fn               [ev=4] Peer is now: 239.255.255.250
2022-01-07 08:37:35 2 main.c:20:tfn              Sending M-SEARCH
2022-01-07 08:37:35 2 main.c:14:fn               [ev=7] Peer is now: 239.255.255.250
2022-01-07 08:37:35 2 main.c:14:fn               [ev=2] Peer is now: 239.255.255.250
2022-01-07 08:37:35 2 main.c:14:fn               [ev=6] Peer is now: 192.168.1.1  << Data received from remote
2022-01-07 08:37:35 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
.
.
.
2022-01-07 08:37:36 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:36 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:36 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:37 2 main.c:20:tfn              Sending M-SEARCH
2022-01-07 08:37:37 2 main.c:14:fn               [ev=7] Peer is now: 192.168.1.1  << SEARCH has gone to this IP
2022-01-07 08:37:37 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:37 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:37 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:37 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:37 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:38 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
2022-01-07 08:37:38 2 main.c:14:fn               [ev=2] Peer is now: 192.168.1.1
```

Before this change, I had it working by setting the SO_REUSEADDR on the mg_connection FD (available after calling `mg_connect(...)`, which then allowed the same port number to be opened on the interface as a new listener (responders to the SSDP will use the port it received the multicast on).

cpq, 2022-01-07:
@gegzy that's not just you. That is the side-effect of this logic:
https://github.com/cesanta/mongoose/blob/ebda2cb100ebfc0699f354a4a5dae23c6e782842/src/sock.c#L250-L251

Originally, it was done for the listening UDP connections. Since there is no "accept()", a receiving socket should know the remote peer that sent data, in order to be able to respond back.

In our case, this behavior is not desirable. What we can do is to change the condition, from `if (n > 0)` to `if (n > 0 && c->is_listening)` - and in that case, the SSDP socket created by `mg_connect()` call, will not have `c->is_listening` flag set and therefore won't get the `c->peer` changed with every new response. On the other hand, we won't be able to respond back.

The other way is to re-initialise `c->peer` to the multicast address before every new search request using the `mg_resolve()` API function.

gegzy, 2022-01-07:
Thanks @cpq

cpq, 2022-01-07:
Thanks @gegzy and @mill1000 
Closing this issue now.

closed 2022-01-07 by cpq