cesanta/mongoose #1441: Memory error detected in http-revere-proxy example

cyanhill, 2022-01-05:
- My goal is: reverse proxy gSOAP server
- My actions were: used the http-reverse-proxy example and [a simple gSOAP server](https://www.genivia.com/examples/calc/index.html#Implementing_a_stand-alone_server_application) listen port 8081. 
- My expectation was: reverse proxy succeeded
- The result I saw: http-revere-proxy example maybe print `Segmentation fault` or `Abort`. Then I used AddressSanitizer to detect memory errors.
- My question is: Is it the problem caused by the wrong way of using it?

# Environment

- mongoose version: 7.5
- Compiler/IDE and SDK: n/a
- Target hardware/board: n/a
- Connectivity chip/module: n/a

I replaced the url of the http-reverse-proxy example with this snippet:

```c
static const char *s_backend_url = "http://127.0.0.1:8081";
static const char *s_listen_url = "http://0.0.0.0:8080";
```

Log

```
2022-01-05 09:27:51 3 mongoose.c:3353:mg_listen  1 accepting on http://0.0.0.0:8080 (port 8080)
2022-01-05 09:27:52 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 r- tchrc
2022-01-05 09:27:53 3 mongoose.c:3255:accept_con 2 accepted 127.0.0.1:44432
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 r- tchrc
2022-01-05 09:27:53 4 mongoose.c:3087:read_conn  2   00100000000010       0 78/2048 err 0
2022-01-05 09:27:53 3 mongoose.c:3223:mg_connect 3 -> http://127.0.0.1:8081
2022-01-05 09:27:53 3 main.c:32:forward_request  FORWARDING: GET /
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 3 -w tChrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 3 -w tchrc
2022-01-05 09:27:53 4 mongoose.c:3116:write_conn 3   01000000000001      73 73 err 115
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 3 r- tchrc
2022-01-05 09:27:53 4 mongoose.c:3087:read_conn  3   01000000000010       0 640/2048 err 115
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 3 r- tchrc
2022-01-05 09:27:53 3 mongoose.c:3087:read_conn  3   01000000000010       0 -1/2048 err 115
2022-01-05 09:27:53 3 mongoose.c:3144:close_conn 3 closed
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 -w tchrc
2022-01-05 09:27:53 4 mongoose.c:3116:write_conn 2   00100000000001     640 640 err 115
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 1 -- tchrc
2022-01-05 09:27:53 4 mongoose.c:3439:mg_mgr_pol 2 r- tchrc
2022-01-05 09:27:53 3 mongoose.c:3087:read_conn  2   00100000000010       0 -1/2048 err 115
=================================================================
==15014==ERROR: AddressSanitizer: heap-use-after-free on address 0x611000000390 at pc 0x5586286530e1 bp 0x7ffe9896f120 sp 0x7ffe9896f110
WRITE of size 2 at 0x611000000390 thread T0
    #0 0x5586286530e0 in fn /media/tiger/shenyun/test/mg//main.c:63
    #1 0x55862862e20d in mg_call /media/tiger/shenyun/test/mg//mongoose.c:391
    #2 0x558628644987 in close_conn /media/tiger/shenyun/test/mg//mongoose.c:3143
    #3 0x558628648c54 in mg_mgr_poll /media/tiger/shenyun/test/mg//mongoose.c:3455
    #4 0x558628653260 in main /media/tiger/shenyun/test/mg//main.c:74
    #5 0x7fe2c3f8e0b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)
    #6 0x55862852c69d in _start (/media/tiger/shenyun/test/mg/build/linux/_install/mg+0xf69d)

0x611000000390 is located 208 bytes inside of 216-byte region [0x6110000002c0,0x611000000398)
freed by thread T0 here:
    #0 0x5586285ea0d8 in free (/media/tiger/shenyun/test/mg/build/linux/_install/mg+0xcd0d8)
    #1 0x558628644aa9 in close_conn /media/tiger/shenyun/test/mg//mongoose.c:3156
    #2 0x558628648c54 in mg_mgr_poll /media/tiger/shenyun/test/mg//mongoose.c:3455
    #3 0x558628653260 in main /media/tiger/shenyun/test/mg//main.c:74
    #4 0x7fe2c3f8e0b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)

previously allocated by thread T0 here:
    #0 0x5586285ea658 in calloc (/media/tiger/shenyun/test/mg/build/linux/_install/mg+0xcd658)
    #1 0x5586286428de in alloc_conn /media/tiger/shenyun/test/mg//mongoose.c:2943
    #2 0x5586286454c5 in mg_connect /media/tiger/shenyun/test/mg//mongoose.c:3214
    #3 0x558628652ef5 in fn /media/tiger/shenyun/test/mg//main.c:50
    #4 0x55862862e20d in mg_call /media/tiger/shenyun/test/mg//mongoose.c:391
    #5 0x5586286387c0 in http_cb /media/tiger/shenyun/test/mg//mongoose.c:1684
    #6 0x55862862e28d in mg_call /media/tiger/shenyun/test/mg//mongoose.c:392
    #7 0x558628643f73 in read_conn /media/tiger/shenyun/test/mg//mongoose.c:3100
    #8 0x558628648a53 in mg_mgr_poll /media/tiger/shenyun/test/mg//mongoose.c:3449
    #9 0x558628653260 in main /media/tiger/shenyun/test/mg//main.c:74
    #10 0x7fe2c3f8e0b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)

SUMMARY: AddressSanitizer: heap-use-after-free /media/tiger/shenyun/test/mg//main.c:63 in fn
Shadow bytes around the buggy address:
  0x0c227fff8020: 00 00 00 fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff8030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c227fff8040: 00 00 00 00 00 00 00 00 00 00 00 fa fa fa fa fa
  0x0c227fff8050: fa fa fa fa fa fa fa fa fd fd fd fd fd fd fd fd
  0x0c227fff8060: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
=>0x0c227fff8070: fd fd[fd]fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff8080: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff8090: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff80a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff80b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c227fff80c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==15014==ABORTING
```

`curl http://127.0.0.1:8080`does return after sending the request. But AddressSanitizer detected a memory error.

cpq, 2022-01-05:
Pushed https://github.com/cesanta/mongoose/commit/6ad4d242c2a25e655bd14507cd446a5d20120249 - please pull and retry.

cyanhill, 2022-01-06:
> Pushed [6ad4d24](https://github.com/cesanta/mongoose/commit/6ad4d242c2a25e655bd14507cd446a5d20120249) - please pull and retry.

Great! The reverse proxy is currently working well.

cpq, 2022-01-06:
Thanks. Closing this.

closed 2022-01-06 by cpq