cesanta/mongoose #1446: Upload file on ESP32 doesn't work

SgtPepperFTW, 2022-01-11:
- My goal is: Upload file using file-upload example on ESP32
- My actions were: repeat example from file-upload on ESP32
- My expectation was: File uploaded in /tmp
- The result I saw: 

1970-01-01 00:00:21 3 mongoose.c:3274:accept_con 2 accepted 192.168.1.127:62614
1970-01-01 00:00:27 3 mongoose.c:1158:mg_http_up 0x37 44 bytes @ 0 [testupload.txt]
1970-01-01 00:00:28 3 mongoose.c:1158:mg_http_up 0x37 0 bytes @ 44 [testupload.txt]
1970-01-01 00:06:48 3 mongoose.c:2979:iolog      2   00100000000010 0:0 -1 err 128
1970-01-01 00:06:48 3 mongoose.c:3162:close_conn 2 closed

Browser page stuck at "sending bytes 44..44 of 44"
- My question is: How to repair this?

# Environment

- mongoose version: 7.5
- Compiler/IDE and SDK: Visual Studio Code + Platformio + Espressif framework
- Target hardware/board: ESP32
- Browsers tried: Chrome, Firefox, Edge

SgtPepperFTW, 2022-01-11:
Ok, so file is actually uploaded, it's just you don't see it, because just calling the server return folder "/spiffs" and the file is save in "/tmp". Still curious, what iolog error is. And should the page be stuck?

cpq, 2022-01-12:
The page should not be stuck.
Closing this, as upload works.

closed 2022-01-12 by cpq