cesanta/mongoose #1453: valgrind complains about recent changes to mg_mgr_wakeup

jvo203, 2022-01-21:
<cont void* buf, size_t len> have recently been added to mg_mgr_wakeup(). Up until these changes were made to mongoose valgrind had been declaring it clean. But following these changes using either ```mg_mgr_wakeup(c, NULL, 0);``` or ```mg_mgr_wakeup(c, "", 0);``` results in the following complaints from valgrind:

==41917== Thread 8:
==41917== Invalid read of size 8
==41917==    at 0x49A600: mg_mgr_wakeup (mongoose.c:3514)
==41917==    by 0x5D7AA86: start_thread (pthread_create.c:435)
==41917==    by 0x5DFE8D3: clone (clone.S:100)
==41917==  Address 0x7124628 is 136 bytes inside a block of size 216 free'd
==41917==    at 0x48440E4: free (vg_replace_malloc.c:872)
==41917==    by 0x4A1340: close_conn (mongoose.c:3369)
==41917==    by 0x4A1340: mg_mgr_poll (mongoose.c:3676)
==41917==    by 0x403D2C: main (main.c:312)
==41917==  Block was alloc'd at
==41917==    at 0x4846464: calloc (vg_replace_malloc.c:1328)
==41917==    by 0x49A7D5: alloc_conn (mongoose.c:3158)
==41917==    by 0x49A7D5: mg_mkpipe (mongoose.c:3534)
==41917==    by 0x403CF5: main (main.c:307)
==41917== 
2022-01-21 06:45:23 2 mongoose.c:3514:mg_mgr_wak skt: 0x7
==41917== Invalid read of size 4
==41917==    at 0x49A615: mg_mgr_wakeup (mongoose.c:3514)
==41917==    by 0x5D7AA86: start_thread (pthread_create.c:435)
==41917==    by 0x5DFE8D3: clone (clone.S:100)
==41917==  Address 0x7124628 is 136 bytes inside a block of size 216 free'd
==41917==    at 0x48440E4: free (vg_replace_malloc.c:872)
==41917==    by 0x4A1340: close_conn (mongoose.c:3369)
==41917==    by 0x4A1340: mg_mgr_poll (mongoose.c:3676)
==41917==    by 0x403D2C: main (main.c:312)
==41917==  Block was alloc'd at
==41917==    at 0x4846464: calloc (vg_replace_malloc.c:1328)
==41917==    by 0x49A7D5: alloc_conn (mongoose.c:3158)
==41917==    by 0x49A7D5: mg_mkpipe (mongoose.c:3534)
==41917==    by 0x403CF5: main (main.c:307)

# Environment

- mongoose version: 7.5
- Compiler/IDE and SDK: gcc (GCC) 11.2.1 20211203 (Red Hat 11.2.1-7)
- Target hardware/board: Fedora release 35, AMD Ryzen 7 3700X 8-Core Processor

jvo203, 2022-01-21:
What is happening is that in
```
void mg_mgr_wakeup(struct mg_connection *c, const void *buf, size_t len) {
  LOG(LL_INFO, ("skt: %p", c->pfn_data));
  send((SOCKET) (size_t) c->pfn_data, (const char *) buf, len, MSG_NONBLOCKING);
}
```
```LOG(LL_INFO, ("skt: %p", c->pfn_data));``` accesses ```c->pfn_data``` *AFTER* c had already been closed and freeded. There must be a way to check inside ```mg_mgr_wakeup()``` whether or not ```c``` is still active.

cpq, 2022-01-21:
Thank you @jvo203 

Since the piped connection is in different task, there is no good way to pass the knowledge about the status of that connection. Normally, it should not get closed.

In your tests with NULL or zero buffer it does get closed, because writing 0 bytes returns 0, and mongoose thinks that `send()` to a socket returned zero and it should be closed. It closes the connection. That should not happen.

A fix is to check for the NULL or zero buffer, and send at least one byte instead.

commit e565a7e7 (2022-01-21): Fix #1453 - do not allow empty wakeup buffer
closed 2022-01-21 by cpq