cesanta/mongoose #1460: Mongoose 7.6 + ESP32 InstrFetchProhibited on serving file (Regression Bug)

pjensen-addco, 2022-01-28:
- My goal is: Not Hardfault when Serving files
- My actions were: Upgrade to MG 7.6 from 7.5 to take advantage of http_upload changes
- My expectation was: Similar stability to 7.5
- The result I saw: 

After upgrading to 7.6 from 7.5, I'm now getting a few hard faults fairly consistently (used to never happen, now happens approx every 2nd or 3rd time)

### Hard fault 1

Serial Output: 
```
[E][webserver.cpp:158] web_handler(): Serving file: /spiffs/socketDispatcher.js
[E][webserver.cpp:158] web_handler(): Serving file: /spiffs/login_scriptfile.js
1970-01-01 00:01:25 1 iobuf.c:36:mg_iobuf_resize 0->2048
1970-01-01 00:01:25 1 iobuf.c:36:mg_iobuf_resize 2048->4096
1970-01-01 00:01:25 1 iobuf.c:36:mg_iobuf_resize 0->4096
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060530  A0      : 0x8017a6bd  A1      : 0x3ffed560  
A2      : 0x3fff17b4  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fff1400  
A6      : 0x3ffc6ddc  A7      : 0xff000000  A8      : 0x800e1bd2  A9      : 0x3ffed540  
A10     : 0x3ffcb618  A11     : 0x3fff59a4  A12     : 0x00001000  A13     : 0x3fff59a4  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4008c854  LEND    : 0x4008c870  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffed560 0x4017a6ba:0x3ffed580 0x400e1f02:0x3ffed5a0 0x400e31d7:0x3ffed6e0 0x400e5e99:0x3ffed780 0x40090e42:0x3ffed7a0
``` 

Decoded Backtrace:
```
Stack trace:
  b'0x4017a6ba: mg_tls_send at C:\\[trimmed]/src/tls_dummy.c:19'
  b'0x400e1f02: mg_error at C:\\[trimmed]/src/event.c:19'
  b'0x400e31d7: mg_mgr_poll at C:\\[trimmed]/src/sock.c:607'
  b'0x400e5e99: my_stat(char const*, unsigned int*, long*) at C:\\[trimmed]/src/webserver.cpp:101'
  b'0x40090e42: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143'
  ```
 Program counter (0x00000000) indicates that it's probably trying to load a function prototype from a null pointer, and failing because of that
 
 **Oddly, I'm not using TLS at all in this application.**
 
 ### Hard Fault 2
 
 Serial Output:

```
 [E][webserver.cpp:158] web_handler(): Serving file: /spiffs/socketDispatcher.js
1970-01-01 00:00:13 1 iobuf.c:36:mg_iobuf_resize 0->2048
1970-01-01 00:00:13 1 event.c:21:mg_error        11 oom
1970-01-01 00:00:13 1 iobuf.c:36:mg_iobuf_resize 2048->4096
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x0000000b  PS      : 0x00060530  A0      : 0x8017a6bd  A1      : 0x3ffed6a0  
A2      : 0x3fff328c  A3      : 0x00000081  A4      : 0x0000000b  A5      : 0x3fff400c  
A6      : 0x00000000  A7      : 0x3ffd98f8  A8      : 0x800e1bd2  A9      : 0x3ffed680  
A10     : 0x3ffcb6e8  A11     : 0x3fff9f0d  A12     : 0x0000077f  A13     : 0x00000000  
A14     : 0x00000800  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000008  LBEG    : 0x400e13c9  LEND    : 0x400e13d6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x0000000b:0x3ffed6a0 0x4017a6ba:0x3ffed6c0 0x400e2f37:0x3ffed6e0 0x400e5e99:0x3ffed780 0x40090e42:0x3ffed7a0
```
Decoded Backtrace:
```
 Stack trace:
  b'0x4017a6ba: mg_call at C:\\[trimmed]/src/event.c:12 (discriminator 1)'
  b'0x400e2f37: mg_mgr_poll at C:\\[trimmed]/src/sock.c:590'
  b'0x400e5e99: WEB_DNS_Server_Task(void*) at C:\\[trimmed]/src/webserver.cpp:382 (discriminator 1)'
  b'0x40090e42: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143'
 ```
Again, program counter of 0x0b appears to indicate loading instructions from an offset off a null pointer, but a different function.
 
If I downgrade to 7.5, the application works perfectly. Note: I do see the ` iobuf.c:36:mg_iobuf_resize 2048->4096 ` errors, but they don't appear to cause issues. They seem to self resolve after a bit and don't cause a hard fault. I have over 66k heap on the internal heap, and 4MB on the external PSRAM, and this is within 30 seconds of boot, so it can't be that fragmented. 
 
This bug is kinda frustrating. I need the custom FS integration with the http_upload (upload to an onboard SD card, rather than spiffs), but the stability is going to keep me stuck on 7.5. 

# Environment

- mongoose version: 7.6
- Compiler/IDE and SDK: PlatformIO+VSCode: Espressif 3.4.0
- Target hardware/board: ESP32 Wrover, 8MB Flash, 8MB PSRAM
- Connectivity chip/module: N/A

pjensen-addco, 2022-01-28:
Only thing that I think that really changed that would impact me is:
`Added ability to override stat() call for POSIX FS` from the changelog, but I haven't really find any documentation on that. Previous code was based off the example, and works just fine.

Maybe this 'Fixed DNS resolution issue: propagating DNS errors to the resolving connection' , but my code doesn't use mg_wakeup, mqtt, http_upload (currently), mg_tls, UDP, or mg_time/usleep.

cpq, 2022-01-28:
Could you symbolize and publish the backtrace please. See https://mdash.net/docs/#crash-backtrace

pjensen-addco, 2022-01-28:
Unless I'm mistaken (which I could be)... The symbolized backtrace from that link looks pretty similar to what I've already posted right below the serial output for the respective hard faults (in the Decoded backtrace). It's off in the weeds in mg_tls_send and mg_call, which is all the stacktrace reports with the result of the outputs of xtensa-esp32-elf-addr2line.exe and xtensa-esp32-elf-gdb.exe.

cpq, 2022-01-29:
It says OOM.
Can you add more traces to your app and track free RAM please?

pjensen-addco, 2022-02-01:
Alright, this is odd. I'm going to be closing this issue, because after adding the RAM trace, I couldn't get it to fail.

I then proceeded to remove the RAM trace from the code..... and it did not go back to failing.

I have not succeeded in my efforts to make it fail in the same way after the update, and seems to be back to stable. 

I think I'm going to write this off as a good old fashioned Friday afternoon bug. Just needed to let the processor think for the weekend about everything it has failed to deliver for me.

Thanks,
-P

closed 2022-02-01 by pjensen-addco