cesanta/mongoose #1462: How to get HTTP status code from response

MFAshby, 2022-02-05:
- My goal is: Be an HTTP client. Check the status code in an HTTP response.
- My actions were: Start with the [example](https://github.com/cesanta/mongoose/blob/master/examples/http-client/main.c) HTTP client code. Look for a function to obtain the HTTP status code as a integer (e.g. 200 OK or 400 Bad Request).
- My expectation was: HTTP response code would be parsed and available in the `struct mg_http_message`, or at least obtainable via a function that accepts `struct mg_http_response` as a parameter.
- The result I saw: There is no such function, and the HTTP status actually ends up as a `char*`in the `->uri` member of the `struct mg_http_message`. 
- My question is: what is the recommended way to check the HTTP response status code using mongoose as an HTTP client? 

at the moment I am obtaining the status code in this way, because it happens that the status code in a response is in the same position as the uri in a request:
```
    // It is unusual that the status code is in the ->uri field.
    char* endPtr = (char*)(hm->uri.ptr+hm->uri.len);
    int statusCode = (int)strtol(hm->uri.ptr, &endPtr, 10);
```

Mongoose appears to be using the same structure for http request and response, which does not accurately model the different structure of HTTP start and status lines. The [start line](https://developer.mozilla.org/en-US/docs/Web/HTTP/Messages#start_line) has a structure like `METHOD REQUEST VERSION` and the [status line](https://developer.mozilla.org/en-US/docs/Web/HTTP/Messages#status_line) has a structure like `VERSION STATUS_CODE STATUS_TEXT`. Suggestion; different structures should be used to represent requests and responses.

# Environment

- mongoose version: 7.6
- Compiler/IDE and SDK: gcc 10.2.0
- Target hardware/board: linux
- Connectivity chip/module: n/a

cpq, 2022-02-08:
Would this helper function solve the issue?

```c
static inline int mg_http_status(const struct mg_http_message *hm) {
   return atoi(hm->uri.ptr);
}
```

MFAshby, 2022-02-08:
@cpq yes it would help me and probably other library users, thanks!

The implementation probably needs a comment because a URI isn't a status code, it could be confusing.

closed 2022-02-12 by cpq