cesanta/mongoose #1466: VULN: requesting server with `GET /.%2f` returns a directory listing, even if it's disabled.

binex-dsk, 2022-02-11:
- My goal is: To completely disable all directory listing
- My actions were: A user entered `/.%2f` into my website using mongoose, https://file.swurl.xyz
- My expectation was: For it to return 404.
- The result I saw: A directory listing, even with dirlist off
- My question is: Why? 

# Environment

- mongoose version: 7.4
- Compiler/IDE and SDK: gcc (GCC) 11.1.0
- Target hardware/board: standard amd64

binex-dsk, 2022-02-11:
I realized it's probably because %2f is /, so it's requesting `./`. However, directory listing should be completely disabled regardless of circumstances.

binex-dsk, 2022-02-12:
I disabled listing by setting `MG_ENABLE_DIRLIST` to 0.

The problem is that `listdir` is not `if`'ed under this macro.

cpq, 2022-02-12:
Yeah, the `MG_ENABLE_DIRLIST` is actually never checked. The code always lists a directory. I'll remove this macro from the code and from the documentation. Please don't use it.
In order to avoid listing a directory, put an empty `index.html` inside.

cpq, 2022-02-12:
You could retry with `-DMG_ENABLE_DIRLIST=0`
That should disable listing on FS level.
Pull recent sources.

binex-dsk, 2022-02-13:
Yeah it doesn't seem to be a problem in the latest commit.

closed 2022-02-13 by binex-dsk