cesanta/mongoose #1467: Mongoose not processing HTTP requests in a non-blocking manner

thooton, 2022-02-14:
Hello, sorry to bother, I have the following code, slightly modified from the tutorial:

```
#pragma comment(lib, "Ws2_32.lib")
#include <iostream>

#include "mongoose.h"

static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
    if (ev == MG_EV_HTTP_MSG) {
        std::cout << "Received HTTP request\n";
        // mg_http_reply(c, 200, "Content-Type: text/plain\r\n", "Hello, %s\n", "world");
    }
}

int main() {
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mg_log_set("3");
    mg_http_listen(&mgr, "0.0.0.0:29906", fn, NULL);              // Create listening connection
    for (;;) mg_mgr_poll(&mgr, 1000);  // Block forever
}
```

I run the code, and send three concurrent HTTP requests. 

Output:

```
2022-02-14 05:44:35 3 mongoose.c:3569:mg_listen  1 accepting on 0.0.0.0:29906 (port 29906)
2022-02-14 05:44:37 3 mongoose.c:3467:accept_con 2 accepted 127.0.0.1:52957
Received message
2022-02-14 05:44:38 3 mongoose.c:3467:accept_con 3 accepted 127.0.0.1:52958
2022-02-14 05:44:44 3 mongoose.c:3170:iolog      2   00100000000010 0:0 -1 err 0
2022-02-14 05:44:44 3 mongoose.c:3353:close_conn 2 closed
Received message
2022-02-14 05:44:45 3 mongoose.c:3170:iolog      3   00100000000010 0:0 -1 err 0
2022-02-14 05:44:45 3 mongoose.c:3353:close_conn 3 closed
2022-02-14 05:44:45 3 mongoose.c:3467:accept_con 4 accepted 127.0.0.1:52961
Received message
2022-02-14 05:44:45 3 mongoose.c:3467:accept_con 5 accepted 127.0.0.1:52962
2022-02-14 05:44:47 3 mongoose.c:3170:iolog      4   00100000000010 0:0 -1 err 0
2022-02-14 05:44:47 3 mongoose.c:3353:close_conn 4 closed
```

Mongoose is processing the requests one by one -- not asynchronous or non-blocking -- am I doing something wrong?

cpq, 2022-02-19:
@thooton please elaborate - what exactly do you expect to see in the "asynchronous and non-blocking" case?

closed 2022-02-19 by cpq

thooton, 2022-03-18:
> Please do you homework well before making statements like you did.

If after hours of attempting to read through the documentation and research in order to figure out how to use this library I am still unable to accomplish a simple task, then I would say this is the fault of the developer, not the user.

What is clear to you may not be to those who have been newly introduced to your work. So if you would care to help out with this, it would be greatly appreciated.

> please elaborate - what exactly do you expect to see in the "asynchronous and non-blocking" case?

I notice that when the HTTP requests are sent to different paths (`http://localhost:29906/a`, `http://localhost:29906/b`, `http://localhost/29906/c`) then I receive the desired behavior, where Mongoose processes the requests individually and does not wait for another to time out:

```2c415860 3 mongoose.c:3615:mg_listen    1 0.0.0.0:29906 port 29906
2c4207ea 3 mongoose.c:3514:accept_conn  2 accepted 127.0.0.1:30747
Received HTTP request
2c4208e4 3 mongoose.c:3514:accept_conn  3 accepted 127.0.0.1:30749
Received HTTP request
2c4208f3 3 mongoose.c:3514:accept_conn  4 accepted 127.0.0.1:30750
Received HTTP request
2c4209ce 3 mongoose.c:3514:accept_conn  5 accepted 127.0.0.1:30751
2c422229 3 mongoose.c:3217:iolog          2 00100000000010 0:0 -1 err 0 (No error)
2c422229 3 mongoose.c:3400:close_conn   2 closed
2c422313 3 mongoose.c:3217:iolog          3 00100000000010 0:0 -1 err 0 (No error)
2c422313 3 mongoose.c:3400:close_conn   3 closed
2c4223fd 3 mongoose.c:3217:iolog          4 00100000000010 0:0 -1 err 0 (No error)
2c4223fd 3 mongoose.c:3400:close_conn   4 closed
```

Do you know how I can replicate this behavior with just one path (`http://localhost:29906`)?

cpq, 2022-03-18:
@thooton it is not our job to educate you.

thooton, 2022-03-19:
@cpq Got it. It's clear that your statement that "Mongoose makes embedded network programming fast, robust, and easy" was a lie. One would think that the developers of a networking library so widely used would make an effort to be a little less condescending to their users.